bnjmnp/pysoem #104: Cannot read device object dictionary

jakezimmer, 2023-04-12:
Hello,

I'm trying to read the object dictionary of a slave.

```python
master = pysoem.Master()

master.open(adapter)

if master.config_init() > 0:
    for slave in master.slaves:
        print(slave.name)
        print(slave.state)
        print(slave.sdo_read(0x1008, 0).decode('utf-8'))
        print(slave.od)
else:
    print('no device found')

master.close()
```
Unfortunately, while I can connect to the slave and read some basic information, I am unable to get the object dictionary.
```bash
EPOS4
1
EPOS4
---------------------------------------------------------------------------
SdoInfoError                              Traceback (most recent call last)
Cell In[14], line 12
     10         print(slave.state)
     11         print(slave.sdo_read(0x1008, 0).decode('utf-8'))
---> 12         print(slave.od)
     13 else:
     14     print('no device found')

File pysoem\pysoem.pyx:1012, in pysoem.pysoem.CdefSlave._get_od()

SdoInfoError: Sdo List Info read failed
```
**Is there anything I'm doing wrong here or other steps needed before reading the object dictionary to get the list of objects?**

> Additional information:
> - OS: Windows 10
> - Slave Device: Maxon EPOS4 EtherCAT Motor Position Controller
> - Can successfully view the object dictionary through a GUI supplied by Maxon

tdemierr, 2023-04-15:
Hello,
I am relatively new to ethercat in with pysoem. However, object dictionary does not seem to be accessible in https://pysoem.readthedocs.io/en/latest/cdef_slave.html# . 

If you want to get values of sdo, you can try to use the https://github.com/bnjmnp/pysoem/blob/master/examples/read_sdo_info.py and adapt it with sdo_read to get the values you are looking for.
Best regards
Tim

Michaelicious, 2025-05-14:
slave.od is accessible after sending an SDO, IF everything is ok.
When I had issues with slave.od, it was due to missing/wrong EEPROM data
1. you need to create a bin file from the EDS file with twinCAT
2. Write eeprom
3. Recover slaves/config_init again. 

should work.