bnjmnp/pysoem #106: ConnectionError: could not open interface

mpratikk, 2023-06-15:
I have listed adapter from my PC using cmd ipconfig/all and copy the description of interface and put it in pysoem code for connection to EtherCAT network.

Even after trying out multiple combination I am getting same issue, Can anyone help me out how can I do this?
I am having adapter:  Intel(R) Ethernet Connection (14) I219-LM

Please let me know If I am doing anything wrong.

bnjmnp, 2023-06-20:
What operating system are you using? Try to run the [examples/find_adapters.py](https://github.com/bnjmnp/pysoem/blob/ad38e0b49f2501181742740ab05c1752f0f4a13d/examples/find_adapters.py). You will get a list of adapter IDs. Then just copy the ID of the desired adapter to the open() function.

mpratikk, 2023-06-20:
I am using Windows 11, I have tested with find_adapters.py and added the adapter as below
Adapter 4
  \Device\NPF_{AC935391-4A48-48D2-A544-44270B4D5C31}
  Intel(R) Ethernet Connection (14) I219-LM

Do we only need to use id as 'AC935391-4A48-48D2-A544-44270B4D5C31' ?

>

bnjmnp, 2023-06-20:
No, everything from the first backslash to the closing curly bracket.

mpratikk, 2023-06-20:
> No, everything from the first backslash to the closing curly brace.

Thank You so much, It is working now

master.open(r'\Device\NPF_{Adapter_ID}')

closed 2023-06-20 by mpratikk