bnjmnp/pysoem #107: Failed to go to OP state for ED3L

Xubunt, 2023-06-16:
Hello, I am trying to move my ESTUN ED3L motor using modified code from issue #30. I am using the code below:
`import pysoem
import time
import ctypes

import threading

ed3l = None

class InputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('statusword_1', ctypes.c_uint16),
        ('modes_of_operation_display_1', ctypes.c_int8),
        ('position_actual_value_1', ctypes.c_int32),
        ('velocity_actual_value_1', ctypes.c_int32),
        ('statusword_2', ctypes.c_uint16),
        ('modes_of_operation_display_2', ctypes.c_int8),
        ('position_actual_value_2', ctypes.c_int32),
        ('velocity_actual_value_2', ctypes.c_int32),
        ('statusword_3', ctypes.c_uint16),
        ('modes_of_operation_display_3', ctypes.c_int8),
        ('position_actual_value_3', ctypes.c_int32),
        ('velocity_actual_value_3', ctypes.c_int32),
    ]

class OutputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('controlword_1', ctypes.c_int16),
        ('modes_of_operation_1', ctypes.c_uint8),
        ('target_position_1', ctypes.c_int32),
        ('target_velocity_1', ctypes.c_int32),
        ('controlword_2', ctypes.c_int16),
        ('modes_of_operation_2', ctypes.c_uint8),
        ('target_position_2', ctypes.c_int32),
        ('target_velocity_2', ctypes.c_int32),
        ('controlword_3', ctypes.c_int16),
        ('modes_of_operation_3', ctypes.c_uint8),
        ('target_position_3', ctypes.c_int32),
        ('target_velocity_3', ctypes.c_int32),
    ]

modes_of_operation = {
    'No mode': 0,
    'Profile position mode': 1,
    'Profile velocity mode': 3,
    'Homing mode': 6,
    'Cyclic synchronous position mode': 8,
    'Cyclic synchronous velocity mode': 9,
    'Cyclic synchronous torque mode': 10,
}

def convert_input_data(data):
    return InputPdo.from_buffer_copy(data)

pd_thread_stop_event = threading.Event()
master = pysoem.Master()
master.open('eth1')
actual_wkc = 0
def ed3l_config_func(slave_pos):
    global ed3l
    # limit maximum current
    #ed3l.sdo_write(0x3491, 0, (2000).to_bytes(length=4, byteorder='little', signed=False))
    # # set torque control P value
    # ed3l.sdo_write(0x2041, 1, (50).to_bytes(length=2, byteorder='little', signed=False))
    # # set torque control I value
    # ed3l.sdo_write(0x2041, 2, (20).to_bytes(length=2, byteorder='little', signed=False))
    # # set motor type = BLDC
    # ed3l.sdo_write(0x2050, 0, bytes([3]))
    # # set commutation mode = Hall sensors
    # ed3l.sdo_write(0x2055, 0, bytes([2]))
    # # inverse hall direction
    # ed3l.sdo_write(0x2070, 2, bytes([1]))
    # # tune hall PHI_E offset
    # ed3l.sdo_write(0x2070, 4, (10500).to_bytes(length=2, byteorder='little', signed=True))

def processdata_thread():
    global master  # not sure if this is necessary
    global pd_thread_stop_event  # not sure if this is necessary
    global actual_wkc  # not sure if this is necessary
    while not pd_thread_stop_event.is_set():
        master.send_processdata()
        actual_wkc = master.receive_processdata(10000)
        if not actual_wkc == master.expected_wkc:
            print('incorrect wkc')
        time.sleep(0.01)

def main():
    global ed3l
    master = pysoem.Master()
    master.open('eth1')  # someting like '\\Device\\NPF_{B4B7A38F-7DEB-43AF-B787B7-EABADE43978EA}' under Windows
    if master.config_init() > 0:
        ed3l = master.slaves[0]
        ed3l.config_func = ed3l_config_func
        master.config_map()
        if master.state_check(pysoem.SAFEOP_STATE, 50_000) == pysoem.SAFEOP_STATE:
            master.state = pysoem.OP_STATE
            proc_thread = threading.Thread(target=processdata_thread)
            proc_thread.start()

            master.send_processdata()  # this is actually done in the "processdata_thread" - maybe it is not needed here
            master.receive_processdata(
                2000)  # this is actually done in the "processdata_thread" - maybe it is not needed here

            master.write_state()
            master.state_check(pysoem.OP_STATE, 5_000_000)
            if master.state == pysoem.OP_STATE:
                output_data = OutputPdo()
                output_data.modes_of_operation_1 = modes_of_operation['Profile velocity mode']
                output_data.target_velocity_1 = 500  # RPM
                for control_cmd in [6, 7, 15]:
                    output_data.controlword_1 = control_cmd
                    ed3l.output = bytes(output_data)  # that is the actual change of the PDO output data
                    master.send_processdata()
                    master.receive_processdata(1_000)
                    time.sleep(0.01)
                try:
                    while 1:
                        master.send_processdata()
                        master.receive_processdata(1_000)
                        print(convert_input_data(ed3l.input).statusword)
                        time.sleep(0.01)
                except KeyboardInterrupt:
                    print('stopped')
                # zero everything
                ed3l.output = bytes(len(ed3l.output))
                master.send_processdata()
                master.receive_processdata(1_000)
            else:
                print('failed to got to op state')
        else:
            print('failed to got to safeop state')
        master.state = pysoem.PREOP_STATE
        master.write_state()
    else:
        print('no device found')
    master.close()

if __name__ == '__main__':
    main()`

I first ran this code without running a process data loop at the beginning and got the `Failed to go to OP state` error. After adding the process data loop that I saw in issue #37 I got the `Value Error: Buffer size too small (should be 6 instead of at least 33 bytes)` and then got `incorrect wkc`

What am I missing? The manual and .xml are attached.

Xubunt, 2023-06-16:
Manual:
https://en.estun.com/static/upload/file/20220719/1658199238719685.pdf

XML

`

-<RxPdo Fixed="false">

<Index>#x1600</Index>

<Name LcId="1033"> Receive PDO Mapping 1</Name>

<Exclude>#x1601</Exclude>

<Exclude>#x1602</Exclude>

<Exclude>#x1603</Exclude>

-<Entry>

<Index>#x6040</Index>

<SubIndex>0</SubIndex>

<BitLen>16</BitLen>

<Name LcId="1033">Controlword</Name>

<DataType>UINT</DataType>

</Entry>

-<Entry>

<Index>#x607A</Index>

<SubIndex>0</SubIndex>

<BitLen>32</BitLen>

<Name LcId="1033">Target position</Name>

<DataType>DINT</DataType>

</Entry>

-<Entry>

<Index>#x60FF</Index>

<SubIndex>0</SubIndex>

<BitLen>32</BitLen>

<Name LcId="1033">Target velocity</Name>

<DataType>DINT</DataType>

</Entry>

-<Entry>

<Index>#x6071</Index>

<SubIndex>0</SubIndex>

<BitLen>16</BitLen>

<Name LcId="1033">Target torque</Name>
[... 871 lines cut ...]
<CycleTime Factor="1">0</CycleTime>

<ShiftTime MinBeforeFrame="100">0</ShiftTime>

</Sm>

</OpMode>

</Dc>

-<Eeprom>

<ByteSize>65536</ByteSize>

<ConfigData>060001CCE8030000</ConfigData>

</Eeprom>

</Device>

</Devices>

</Descriptions>

</EtherCATInfo>`

Xubunt, 2023-06-16:
I ran `minimal example.py` and got the following status code: 
```
1 slaves found and configured
Drive did not reach SAFEOP state
al status code 0x1d (Invalid output configuration)
not all slaves reached SAFEOP state

```

Xubunt, 2023-06-16:
This was my modified code for `minimal example.py`
```
import sys
import struct
import time
import collections

import pysoem

class MinimalExample:

    ESTUN_VENDOR_ID = 0x0000060A
    ED3L_PRODUCT_CODE = 0xED310001

    def __init__(self, ifname):
        self._ifname = ifname
        self._master = pysoem.Master()
        SlaveSet = collections.namedtuple(
            'SlaveSet', 'slave_name product_code config_func')
        self._expected_slave_mapping = {0: SlaveSet('ED3L', self.ED3L_PRODUCT_CODE, self.ed3l_setup)}

    def ed3l_setup(self, slave_pos):
        slave = self._master.slaves[slave_pos]

        slave.sdo_write(0x1c12, 0, struct.pack('B', 0))

        map_1c13_bytes = struct.pack('BxHH', 2, 0x1A01, 0x1A03)
        slave.sdo_write(0x1c13, 0, map_1c13_bytes, True)

    def run(self):

        self._master.open(self._ifname)

        # config_init returns the number of slaves found
        if self._master.config_init() > 0:

            print("{} slaves found and configured".format(
                len(self._master.slaves)))

            for i, slave in enumerate(self._master.slaves):
                assert(slave.man == self.ESTUN_VENDOR_ID)
                assert(
                    slave.id == self._expected_slave_mapping[i].product_code)
                slave.config_func = self._expected_slave_mapping[i].config_func

            # PREOP_STATE to SAFEOP_STATE request - each slave's config_func is called
            self._master.config_map()

            # wait 50 ms for all slaves to reach SAFE_OP state
            if self._master.state_check(pysoem.SAFEOP_STATE, 50000) != pysoem.SAFEOP_STATE:
                self._master.read_state()
                for slave in self._master.slaves:
                    if not slave.state == pysoem.SAFEOP_STATE:
                        print('{} did not reach SAFEOP state'.format(slave.name))
                        print('al status code {} ({})'.format(hex(slave.al_status),
                                                              pysoem.al_status_code_to_string(slave.al_status)))
                raise Exception('not all slaves reached SAFEOP state')

            self._master.state = pysoem.OP_STATE
            self._master.write_state()

            self._master.state_check(pysoem.OP_STATE, 50000)
            if self._master.state != pysoem.OP_STATE:
                self._master.read_state()
                for slave in self._master.slaves:
                    if not slave.state == pysoem.OP_STATE:
                        print('{} did not reach OP state'.format(slave.name))
                        print('al status code {} ({})'.format(hex(slave.al_status),
                                                              pysoem.al_status_code_to_string(slave.al_status)))
                raise Exception('not all slaves reached OP state')

            try:
                while 1:
                    # free run cycle
                    self._master.send_processdata()
                    self._master.receive_processdata(2000)

                    volgage_ch_1_ed3l_as_bytes = self._master.slaves[1].input
                    volgage_ch_1_ed3l_as_int16 = struct.unpack(
                        'hh', volgage_ch_1_ed3l_as_bytes)[0]
                    voltage = volgage_ch_1_ed3l_as_int16 * 10 / 0x8000
                    print('Ed3l Ch 1 PDO: {:#06x}; Voltage: {:.4}'.format(
                        volgage_ch_1_ed3l_as_int16, voltage))

                    time.sleep(1)

            except KeyboardInterrupt:
                # ctrl-C abort handling
                print('stopped')

            self._master.state = pysoem.INIT_STATE
            # request INIT state for all slaves
            self._master.write_state()
            self._master.close()
        else:
            print('slaves not found')

if __name__ == '__main__':

    print('minimal_example')

    if len(sys.argv) > 1:
        try:
            MinimalExample(sys.argv[1]).run()
        except Exception as expt:
            print(expt)
            sys.exit(1)
    else:
        print('usage: minimal_example ifname')
        sys.exit(1)
```

bnjmnp, 2023-06-26:
> Manual: https://en.estun.com/static/upload/file/20220719/1658199238719685.pdf
> 
> @bnjmnp would appreciate any guidance

This site dose not use https, which is strange.

Does the vendor also provide a ESI XML file for this device? There one might find what needs to go into the `ed3l_setup()` function to get rid of the "Invalid output configuration" error.
Where you able to operate the device with any other EtherCAT master system?

Xubunt, 2023-06-26:
[1658199238719685.pdf](https://github.com/bnjmnp/pysoem/files/11873422/1658199238719685.pdf)

Xubunt, 2023-06-26:
[ESTUN_ED3L_V1_00B2-软件.txt](https://github.com/bnjmnp/pysoem/files/11873470/ESTUN_ED3L_V1_00B2-.txt)
Here is the .xml file I got from the vendor. I am not sure I know enough about Ethercat to identify what would need to be added for the correct configuration. 

I was able to operate the device using their GUI called ESView V4, so the drive does work manually and with non-Ethercat comms.

bnjmnp, 2023-06-27:
To get a motor running in PV mode you could try adapting your first code with this:

```py
def ed3l_config_func(slave_pos):
    global ed3l

    map_1c12_bytes = struct.pack('BxH', 1, 0x1600)
    ed3l.sdo_write(0x1c12, 0, map_1c12_bytes, True)

    map_1c13_bytes = struct.pack('BxHH', 2, 0x1A00, 0x1A03)
    ed3l.sdo_write(0x1c13, 0, map_1c13_bytes, True)
```

```py
class OutputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('controlword', ctypes.c_int16),
        ('target_velocity', ctypes.c_int32),
    ]
```

```py
class InputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('statusword', ctypes.c_uint16),
        ('velocity_actual_value', ctypes.c_int32),
        ('torque_actual_value', ctypes.c_int16),
    ]
```

````py
            ...
            if master.state == pysoem.OP_STATE:
                output_data = OutputPdo()
                output_data.target_velocity = 500  # RPM
                for control_cmd in [6, 7, 15]:
                    output_data.controlword = control_cmd
                    ed3l.output = bytes(output_data)  # that is the actual change of the PDO output data
                    master.send_processdata()
                    master.receive_processdata(1_000)
                    time.sleep(0.01)
                try:
                    while 1:
                        master.send_processdata()
                        master.receive_processdata(1_000)
                        print(convert_input_data(ed3l.input).statusword)
                        time.sleep(0.01)
                except KeyboardInterrupt:
                    print('stopped')
                # zero everything
                ed3l.output = bytes(len(ed3l.output))
                master.send_processdata()
                master.receive_processdata(1_000)
            else:
                print('failed to got to op state')
            ...
```

Xubunt, 2023-06-27:
After running the code with your changes, I get `ValueError: Buffer size too small (6 instead of at least 8 bytes)
`
It seems this is an issue with specifying the size of the bytes when using the struct module?

bnjmnp, 2023-06-27:
I did a slight update of my code from the last comment. Not sure if it helps. Could you please post the complete stack trace of the error?

Xubunt, 2023-06-27:
Re-ran and still got same error. 
``` Traceback (most recent call last):
  File "/home/teslauser/git/tesla-aurus/aurus/helpers/closures/Ethercat_communication.py", line 118, in <module>
    main()
  File "/home/teslauser/git/tesla-aurus/aurus/helpers/closures/Ethercat_communication.py", line 98, in main
    print(convert_input_data(ed3l.input).statusword)
  File "/home/teslauser/git/tesla-aurus/aurus/helpers/closures/Ethercat_communication.py", line 39, in convert_input_data
    return InputPdo.from_buffer_copy(data)
ValueError: Buffer size too small (6 instead of at least 8 bytes)
```

Xubunt, 2023-06-28:
When I run the code while the motor is hooked up to ESView, I get this error: 
[image]

bnjmnp, 2023-06-28:
That is strange, you could try to cut the input PDO struct to 6 byte:
```py
class InputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('statusword', ctypes.c_uint16),
        ('velocity_actual_value', ctypes.c_int32),
    ]
```
According to the ESI file it should have 8 byte.

Xubunt, 2023-06-28:
That got rid of the buffer error, thanks! However, I now have another issue. When I run the code with the process data loop, I get this error:
```
incorrect wkc
incorrect wkc
incorrect wkc
1560
incorrect wkc
1560
incorrect wkc
1560
incorrect wkc
1560
incorrect wkc
1560
incorrect wkc
1560
```

When I run the code without the process data loop, I get this error:
```
failed to got to op state
```

For reference, the code with the process data loop is in my original post, the first code. 
The code without the process data loop is below:
```
import struct

import pysoem
import time
import ctypes

import threading

ed3l = None

class InputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('statusword', ctypes.c_uint16),
        ('velocity_actual_value', ctypes.c_int32),
    ]

class OutputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('controlword', ctypes.c_int16),
        ('target_velocity', ctypes.c_int32),
    ]

modes_of_operation = {
    'No mode': 0,
    'Profile position mode': 1,
    'Profile velocity mode': 3,
    'Homing mode': 6,
    'Cyclic synchronous position mode': 8,
    'Cyclic synchronous velocity mode': 9,
    'Cyclic synchronous torque mode': 10,
}

def convert_input_data(data):
    return InputPdo.from_buffer_copy(data)

pd_thread_stop_event = threading.Event()
master = pysoem.Master()
master.open('eth1')
actual_wkc = 0
def ed3l_config_func(slave_pos):
    global ed3l

    map_1c12_bytes = struct.pack('BxH', 1, 0x1600)
    ed3l.sdo_write(0x1c12, 0, map_1c12_bytes, True)

    map_1c13_bytes = struct.pack('BxH', 1, 0x1A00)
    ed3l.sdo_write(0x1c13, 0, map_1c13_bytes, True)

def processdata_thread():
    global master  # not sure if this is necessary
    global pd_thread_stop_event  # not sure if this is necessary
    global actual_wkc  # not sure if this is necessary
    while not pd_thread_stop_event.is_set():
        master.send_processdata()
        actual_wkc = master.receive_processdata(10000)
        if not actual_wkc == master.expected_wkc:
            print('incorrect wkc')
        time.sleep(0.01)

def main():
    global ed3l
    master = pysoem.Master()
    master.open('eth1')  # someting like '\\Device\\NPF_{B4B7A38F-7DEB-43AF-B787B7-EABADE43978EA}' under Windows
    if master.config_init() > 0:
        ed3l = master.slaves[0]
        ed3l.config_func = ed3l_config_func
        master.config_map()
        if master.state_check(pysoem.SAFEOP_STATE, 50_000) == pysoem.SAFEOP_STATE:
            master.state = pysoem.OP_STATE
            proc_thread = threading.Thread(target=processdata_thread)
            proc_thread.start()

            master.send_processdata()  # this is actually done in the "processdata_thread" - maybe it is not needed here
            master.receive_processdata(
                2000)  # this is actually done in the "processdata_thread" - maybe it is not needed here

            master.write_state()
            master.state_check(pysoem.OP_STATE, 5_000_000)
            if master.state == pysoem.OP_STATE:
                output_data = OutputPdo()
                output_data.target_velocity = 500  # RPM
                for control_cmd in [6, 7, 15]:
                    output_data.controlword = control_cmd
                    ed3l.output = bytes(output_data)  # that is the actual change of the PDO output data
                    master.send_processdata()
                    master.receive_processdata(1_000)
                    time.sleep(0.01)
                try:
                    while 1:
                        master.send_processdata()
                        master.receive_processdata(1_000)
                        print(convert_input_data(ed3l.input).statusword)
                        time.sleep(0.01)
                except KeyboardInterrupt:
                    print('stopped')
                # zero everything
                ed3l.output = bytes(len(ed3l.output))
                master.send_processdata()
                master.receive_processdata(1_000)
            else:
                print('failed to got to op state')
        else:
            print('failed to got to safeop state')
        master.state = pysoem.PREOP_STATE
        master.write_state()
    else:
        print('no device found')
    master.close()

if __name__ == '__main__':
    main()
```

bnjmnp, 2023-06-28:
Could be timing issues and maybe the drive you are using checks if process data is coming in precisely. You should also check in the main loop if the drive stays in OP_STATE or falls back to SAFEOP_STATE.

The thing is also that your device needs process data communication running (in parallel) wile going to OP_STATE, thus you cannot really get rid of the processdata_thread().

One thing you could try is, to increase the timeout given to receive_processdata(), maybe 100000 instead of 10000.

Xubunt, 2023-06-28:
I upped to timeout, but still get the error:
```
incorrect wkc
In SAFEOP state
incorrect wkc
incorrect wkc
incorrect wkc
1587
incorrect wkc
1587
incorrect wkc
1587
incorrect wkc
1587
incorrect wkc
1587
incorrect wkc
1587
incorrect wkc
1587
incorrect wkc
1587
incorrect wkc
1587
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
1591
incorrect wkc
^Cstopped
In OP state
incorrect wkc
incorrect wkc
incorrect wkc
incorrect wkc
incorrect wkc
incorrect wkc
```

Xubunt, 2023-07-03:
Tried adding ```py global master``` to main and now no longer get `incorrect wkc`. However, the motor does not move and every time I run the script I get the Overtime Watchdog error I mentioned above. 
By the way, what do the numbers that constantly print to terminal mean? The 1591, 1587, 1544 etc

Xubunt, 2023-07-05:
I realize the numbers are the statuswords. What I am confused about it that the system is in OP state and the statusword is printing to terminal, but the motor is not moving. And it is because of the Overtime Watchdog error. Not sure if I can add a time.sleep somewhere to help the system catch up with the code or not - otherwise I am unsure of what to do

bnjmnp, 2023-07-06:
Looks like your device requires a more rigorous/precise real time system. You could rather try to decrease the sleep of 0.01 after the `master.receive_processdata(1_000)` to 0.001 or removing it. You should also remove the print of the statusword, this slows down the loop a lot.

bnjmnp, 2023-07-06:
Another thing you could try is to increase the watchdog timeouts with the set_watchdog() function. Check out #71, maybe that helps.

Xubunt, 2023-07-06:
I did a few tweaks to the sleep and the watchdog error is gone. I get no alarms for when I run the script, but now the motor has an overtravel status when I run the script and it does not move. I have the force stop the script otherwise it keeps running. Here is the stack trace:
` ` `py
sudo python3 Ethercat_communication.py 
In SAFEOP State
In OP state
Made it this far
0x631
Made it this far
0x633
Made it this far
0x633
^Cstopped
Finished OP loop
incorrect wkc
incorrect wkc
^CException ignored in: <module 'threading' from '/usr/lib/python3.10/threading.py'>
incorrect wkc
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1567, in _shutdown
    lock.acquire()
KeyboardInterrupt: 
```
I looked at the watchdog function. What exactly should I include from your code? I currently have this, but it doesn't seem right:
` ` `py
import struct

import pysoem
import time
import ctypes
import pytest
import threading

ed3l = None

class InputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('statusword', ctypes.c_uint16),
        ('velocity_actual_value', ctypes.c_int32),
    ]

class OutputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('controlword', ctypes.c_int16),
        ('target_velocity', ctypes.c_int32),
    ]

modes_of_operation = {
    'No mode': 0,
    'Profile position mode': 1,
    'Profile velocity mode': 3,
    'Homing mode': 6,
    'Cyclic synchronous position mode': 8,
    'Cyclic synchronous velocity mode': 9,
    'Cyclic synchronous torque mode': 10,
}

def convert_input_data(data):
    return InputPdo.from_buffer_copy(data)

pd_thread_stop_event = threading.Event()
master = pysoem.Master()
master.open('eth1')
actual_wkc = 0
def ed3l_config_func(slave_pos):
    global ed3l

    map_1c12_bytes = struct.pack('BxH', 1, 0x1600)
    ed3l.sdo_write(0x1c12, 0, map_1c12_bytes, True)

    map_1c13_bytes = struct.pack('BxH', 1, 0x1A00)
    ed3l.sdo_write(0x1c13, 0, map_1c13_bytes, True)

def processdata_thread():
    global master  # not sure if this is necessary
    global pd_thread_stop_event  # not sure if this is necessary
    global actual_wkc  # not sure if this is necessary
    while not pd_thread_stop_event.is_set():
        master.send_processdata()
        actual_wkc = master.receive_processdata(100000)
        if not actual_wkc == master.expected_wkc:
            print('incorrect wkc')
        time.sleep(0.001)
def test_watchdog_update(eth1, watchdog_register_fix, wd, time_ms, expected_reg_value):
    """Test the set_watchdog() function of the CdefSlave object."""
    wd_reg = {
        'pdi': 0x410,
        'processdata': 0x420,
    }
    if isinstance(expected_reg_value, AttributeError):
        with pytest.raises(AttributeError):
            eth1.set_watchdog(wd_type=wd, wd_time_ms=time_ms)
    else:
        eth1.set_watchdog(wd_type=wd, wd_time_ms=time_ms)
        assert eth1._fprd(address=wd_reg[wd], size=2, timeout_us=4000) == expected_reg_value.to_bytes(2, byteorder='little', signed=False)

def main():
    global ed3l
    global master
    master = pysoem.Master()
    master.open('eth1')  # someting like '\\Device\\NPF_{B4B7A38F-7DEB-43AF-B787B7-EABADE43978EA}' under Windows
    if master.config_init() > 0:
        ed3l = master.slaves[0]
        ed3l.config_func = ed3l_config_func
        master.config_map()
        if master.state_check(pysoem.SAFEOP_STATE, 50_000) == pysoem.SAFEOP_STATE:
            print('In SAFEOP State')
            master.state = pysoem.OP_STATE
            proc_thread = threading.Thread(target=processdata_thread)
            proc_thread.start()

            master.send_processdata()  # this is actually done in the "processdata_thread" - maybe it is not needed here
            master.receive_processdata(
                2000)  # this is actually done in the "processdata_thread" - maybe it is not needed here

            master.write_state()
            master.state_check(pysoem.OP_STATE, 5_000_000)
            if master.state == pysoem.OP_STATE:
                print('In OP state')
                output_data = OutputPdo()
                output_data.target_velocity = 100  # RPM
                for control_cmd in [6, 7, 15]:
                    output_data.controlword = control_cmd
                    ed3l.output = bytes(output_data)  # that is the actual change of the PDO output data
                    master.send_processdata()
                    master.receive_processdata(1_000)
                    print('Made it this far')
                    time.sleep(0.01)
                    print(hex(int.from_bytes(ed3l.sdo_read(0x6041, 0), byteorder='little', signed=False)))
                try:
                    while 1:
                        master.send_processdata()
                        master.receive_processdata(1_000)
                        # print(convert_input_data(ed3l.input).statusword)
                        time.sleep(0.5)
                except KeyboardInterrupt:
                    print('stopped')
                # zero everything
                ed3l.output = bytes(len(ed3l.output))
                master.send_processdata()
                master.receive_processdata(1_000)
                print('Finished OP loop')
            else:
                print('failed to got to op state')
        else:
            print('failed to got to safeop state')
        master.state = pysoem.PREOP_STATE
        master.write_state()
    else:
        print('no device found')
    master.close()

if __name__ == '__main__':
    main()
` ` `

Xubunt, 2023-07-10:
Could it be that I need to initiate homing? If so, do I put that in the control_cmd sequence? In this case it would be 1

Xubunt, 2023-08-03:
@bnjmnp any guidance is appreciated

Xubunt, 2023-08-16:
@bnjmnp Found out how to set the homing through the UI for now. Current status is that drive is in RUN mode, but the motor just won't turn. Here is current draft of code:

```python
import struct

import pysoem
import time
import ctypes
# import pytest
import threading

ed3l = None

class InputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('statusword', ctypes.c_uint16),
        ('velocity_actual_value', ctypes.c_int32),
    ]

class OutputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('controlword', ctypes.c_int16),
        ('target_velocity', ctypes.c_int32),
    ]

modes_of_operation = {
    'No mode': 0,
    'Profile position mode': 1,
    'Profile velocity mode': 3,
    'Homing mode': 6,
    'Cyclic synchronous position mode': 8,
    'Cyclic synchronous velocity mode': 9,
    'Cyclic synchronous torque mode': 10,
}

def convert_input_data(data):
    return InputPdo.from_buffer_copy(data)

pd_thread_stop_event = threading.Event()
master = pysoem.Master()
master.open('eth1')
actual_wkc = 0
def ed3l_config_func(slave_pos):
    global ed3l

    map_1c12_bytes = struct.pack('BxH', 1, 0x1600)
    ed3l.sdo_write(0x1c12, 0, map_1c12_bytes, True)

    map_1c13_bytes = struct.pack('BxH', 1, 0x1A00)
    ed3l.sdo_write(0x1c13, 0, map_1c13_bytes, True)

def processdata_thread():
    global master  # not sure if this is necessary
    global pd_thread_stop_event  # not sure if this is necessary
    global actual_wkc  # not sure if this is necessary
    while not pd_thread_stop_event.is_set():
        master.send_processdata()
        actual_wkc = master.receive_processdata(100000)
        if not actual_wkc == master.expected_wkc:
            print('incorrect wkc')
        time.sleep(0.001)

def main():
    global ed3l
    global master
    master = pysoem.Master()
    master.open('eth1')  # someting like '\\Device\\NPF_{B4B7A38F-7DEB-43AF-B787B7-EABADE43978EA}' under Windows
    if master.config_init() > 0:
        ed3l = master.slaves[0]
        ed3l.config_func = ed3l_config_func
        master.config_map()
        if master.state_check(pysoem.SAFEOP_STATE, 50_000) == pysoem.SAFEOP_STATE:
            print('In SAFEOP State')
            master.state = pysoem.OP_STATE
            proc_thread = threading.Thread(target=processdata_thread)
            proc_thread.start()

            master.send_processdata()  # this is actually done in the "processdata_thread" - maybe it is not needed here
            master.receive_processdata(
                2000)  # this is actually done in the "processdata_thread" - maybe it is not needed here

            master.write_state()
            master.state_check(pysoem.OP_STATE, 5_000_000)
            if master.state == pysoem.OP_STATE:
                print('In OP state')
                output_data = OutputPdo()
                output_data.modes_of_operation_1 = modes_of_operation['Profile velocity mode']
                output_data.target_velocity = 500  # RPM
                for control_cmd in [6, 7, 15]:
                    output_data.controlword = control_cmd
                    ed3l.output = bytes(output_data)  # that is the actual change of the PDO output data
                    master.send_processdata()
                    master.receive_processdata(1_000)
                    time.sleep(0.01)
                    print(hex(int.from_bytes(ed3l.sdo_read(0x6041, 0), byteorder='little', signed=False)))
                try:
                    while 1:
                        master.send_processdata()
                        master.receive_processdata(1_000)
                        print(convert_input_data(ed3l.input).velocity_actual_value)
                        time.sleep(0.005)
                        # print('Made it here')
                except KeyboardInterrupt:
                    print('stopped')
                # zero everything
                ed3l.output = bytes(len(ed3l.output))
                master.send_processdata()
                master.receive_processdata(1_000)
            else:
                print('failed to got to op state')
        else:
            print('failed to got to safeop state')
        master.state = pysoem.PREOP_STATE
        master.write_state()
    else:
        print('no device found')
    master.close()

if __name__ == '__main__':
    main()

closed 2023-08-17 by Xubunt