bnjmnp/pysoem #108: pip install pysoem on Raspbian bullseye failed

finger42, 2023-07-10:
Hello,
I like to use pysoem on Raspberry PIs, but the installation failed:

```
cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```

```
sudo python3 -m pip install pysoem
Looking in indexes: .... 
Collecting pysoem
  Downloading .../pysoem-1.1.0.tar.gz (163 kB)
     |████████████████████████████████| 163 kB 23.7 MB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Building wheels for collected packages: pysoem
  Building wheel for pysoem (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /tmp/tmp_6cqb6wq_in_process.py build_wheel /tmp/tmpfm90d3cc
       cwd: /tmp/pip-install-7sv_4u58/pysoem_1b036e638a9e4b81a7482e49ad0cb05e
  Complete output (22 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-cpython-39
  creating build/lib.linux-armv7l-cpython-39/pysoem
  copying pysoem/__init__.py -> build/lib.linux-armv7l-cpython-39/pysoem
  running build_ext
  building 'pysoem.pysoem' extension
  creating build/temp.linux-armv7l-cpython-39
  creating build/temp.linux-armv7l-cpython-39/soem
  creating build/temp.linux-armv7l-cpython-39/soem/osal
  creating build/temp.linux-armv7l-cpython-39/soem/osal/linux
  creating build/temp.linux-armv7l-cpython-39/soem/oshw
  creating build/temp.linux-armv7l-cpython-39/soem/oshw/linux
  creating build/temp.linux-armv7l-cpython-39/soem/soem
  creating build/temp.linux-armv7l-cpython-39/pysoem
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DEC_VER2= -I./pysoem -I./soem/oshw/linux -I./soem/osal/linux -I./soem/oshw -I./soem/osal -I./soem/soem -I/usr/include/python3.9 -c ./soem/osal/linux/osal.c -o build/temp.linux-armv7l-cpython-39/./soem/osal/linux/osal.o
  arm-linux-gnueabihf-gcc: error: ./soem/osal/linux/osal.c: No such file or directory
  arm-linux-gnueabihf-gcc: fatal error: no input files
  compilation terminated.
  error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
  ----------------------------------------
  ERROR: Failed building wheel for pysoem
Failed to build pysoem
ERROR: Could not build wheels for pysoem which use PEP 517 and cannot be installed directly

```

Any ideas to dig in this problem on my PI ?

closed 2023-07-10 by bnjmnp

bnjmnp, 2023-07-10:
Installs should work again. Shout out to @SR4ven for providing the fix.

finger42, 2023-07-13:
Work like a charm. THX

PR #109 (2023-07-10): Checkout submodules in build_sdist
When building the sdist package the soem submodule was not checked out.
This lead to failing installs.

Fixes #108