bnjmnp/pysoem #112: Using Realtek USB GbE Family Controller for EtherCAT

tomhira, 2023-07-25:
Hello, I started to use new PC, and it has no Ethernet port but usb-type C port.
With it, Realtek USB GbE Family Controller was used via Ethernet port adapter.
It does not work with pysoem as no slave found.

Could someone use it? Does pysoem work with Realtek USB GbE Family Controller?

bnjmnp, 2023-07-26:
Over the time, I used a couple of different USB to Ethernet adapters under Windows (up to W10), they all seemed to work.
You could check if devices are found if you run the command prompt you use to run the Python script with administrator privilege.

tomhira, 2023-07-26:
Hello bnjmnp.
Always thanks for the help.
I run with Python script with administrator privilages but the result is same. No device found.
I am wondering if using DST-C09BK (elecom) with USB Type-C to HDMI, LAN, USB3.0, affect the result?
It is 5Gbps max.

tomhira, 2023-07-28:
Hello bnjmnp
this is the transaction sequence. The program end with no device found.
I compared with the OLD PC of successful transaction.

The only difference seems the IRQ in EtherCAT header.
When successful it is 0x004
For not successful it is 0x000

Do you know how master.config_init() function judges if it found device?
I attach Wireshark capture.
Please check.

[USB_adaptor_debug.zip](https://github.com/bnjmnp/pysoem/files/12190653/USB_adaptor_debug.zip)

tomhira, 2023-08-08:
Hello, it seems the ncap is not properly installed with Windows 10 by selecting the option
Install Npcap in WinPcap API-compatible Mode (WinPcap will be uninstalled)

closed 2023-08-08 by tomhira