bnjmnp/pysoem #113: ARM support

FrederikVanEeckeVintecc, 2023-07-26:
Hi,

I'm trying to pip install pysoem on my jetson orin nx, which is an ARM device. I'm using python3.8.10 and cython 0.29.34. The install fails both locally as within my docker container.

Any ideas what may be causing this error?

Kinds regards,
Frederik

Collecting pysoem
Downloading pysoem-1.1.3.tar.gz (369 kB)
|████████████████████████████████| 369 kB 5.5 MB/s
Installing build dependencies ... done
Getting requirements to build wheel ... error
ERROR: Command errored out with exit status 1:
command: /usr/bin/python3 /tmp/tmpi3sxi59n get_requires_for_build_wheel /tmp/tmph0hok2ok
cwd: /tmp/pip-install-dshh81e8/pysoem
Complete output (59 lines):
warning: pysoem/pysoem.pyx:146:4: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
warning: pysoem/pysoem.pyx:147:4: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
warning: pysoem/pysoem.pyx:148:4: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
warning: pysoem/pysoem.pyx:149:4: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
warning: pysoem/pysoem.pyx:150:4: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
warning: pysoem/pysoem.pyx:151:4: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
warning: pysoem/pysoem.pyx:590:4: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
warning: pysoem/pysoem.pyx:591:4: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
warning: pysoem/cpysoem.pxd:18:4: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
warning: pysoem/cpysoem.pxd:19:4: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
Error compiling Cython file:
...
self._cd.func = value
self._ec_slave.user = <void*>self._cd
if value is None:
self._ec_slave.PO2SOconfig = NULL
else:
self._ec_slave.PO2SOconfig = _xPO2SOconfig
^

pysoem/pysoem.pyx:1013:41: Cannot assign type 'int (uint16, void ) except? -1' to 'int ()(uint16, void *) noexcept'
Error compiling Cython file:
...
self._cd.func = value
self._ec_slave.user = <void*>self._cd
if value is None:
self._ec_slave.PO2SOconfig = NULL
else:
self._ec_slave.PO2SOconfig = _xPO2SOconfigEx
^
pysoem/pysoem.pyx:1021:41: Cannot assign type 'int (uint16, void ) except? -1' to 'int ()(uint16, void *) noexcept'
Compiling pysoem/pysoem.pyx because it changed.
[1/1] Cythonizing pysoem/pysoem.pyx
Traceback (most recent call last):
File "/tmp/tmpi3sxi59n", line 280, in
main()
File "/tmp/tmpi3sxi59n", line 263, in main
json_out['return_val'] = hook(**hook_input['kwargs'])
File "/tmp/tmpi3sxi59n", line 114, in get_requires_for_build_wheel
return hook(config_settings)
File "/tmp/pip-build-env-q7ipx9eh/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 341, in get_requires_for_build_wheel
return self._get_build_requires(config_settings, requirements=['wheel'])
File "/tmp/pip-build-env-q7ipx9eh/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 323, in _get_build_requires
self.run_setup()
File "/tmp/pip-build-env-q7ipx9eh/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 338, in run_setup
exec(code, locals())
File "", line 91, in
File "/tmp/pip-build-env-q7ipx9eh/overlay/lib/python3.8/site-packages/Cython/Build/Dependencies.py", line 1134, in cythonize
cythonize_one(*args)
File "/tmp/pip-build-env-q7ipx9eh/overlay/lib/python3.8/site-packages/Cython/Build/Dependencies.py", line 1301, in cythonize_one
raise CompileError(None, pyx_file)
Cython.Compiler.Errors.CompileError: pysoem/pysoem.pyx

ERROR: Command errored out with exit status 1: /usr/bin/python3 /tmp/tmpi3sxi59n get_requires_for_build_wheel /tmp/tmph0hok2ok Check the logs for full command output.

bnjmnp, 2023-07-27:
I would expect that this is related to the very recently updated Cython package. If you jump to the second last Cython version via `pip install Cython==0.29.36`, the pysoem install might work.

FrederikVanEeckeVintecc, 2023-07-27:
Hi, thanks for the response. Fixing the Cython version to 0.29.36 within the pyproject.toml file solved the problem.

alexVinarskis, 2023-07-31:
I can confirm I got the same problem. Works just fine on `amd64`, but exactly same error on `arm64`/Jetson. Suggestion to `pip install Cython==0.29.36` did not help in my case. Did not try to fix it in source in `pyproject.toml` though.

For now walkaround was downgrading to `1.0.8` of `pysoem`. However, it seems to not always work still, still fails in CI/`arm64v8` docker, but passes natively on Jetson, which is weird, since on purpose both are using same distros (focal, python 3.8.10). I just kept wheel built on Jetson, but thats not a permanent/maintainable solution...

closed 2023-07-31 by bnjmnp

bnjmnp, 2023-07-31:
I tried to fix the compilation error caused by the new Cython version, and I hope the new pysoem version just uploaded PyPI works fine.

alexVinarskis, 2023-08-01:
Thanks for incredibly fast support :)
Confirmed working in [docker arm64v8](https://hub.docker.com/repository/docker/anynameaaa/arm64v8_buildenv/general)

commit 321f2f36 (2023-07-31): Mark callback functions as noexcept to support Cython 3

fix #113