bnjmnp/pysoem #116: Is there way to check the content of the pdo mapping

tomhira, 2023-08-21:
Hello bnjmnp.
Is there way to check the content of the pdo mapping?
I have trouble to know why only 3 out of 4, are mapped.
I can see it with twincat3, but also I would like to check in pysoem.

bnjmnp, 2023-08-21:
You could read out the assign objects 0x1C12 and 0x1C13, and from there the listed PDO mapping objects using SDO read. I found an SOEM C code example that does that: [SOEM/test/linux/slaveinfo/slaveinfo.c#L332C5-L332C15](https://github.com/OpenEtherCATsociety/SOEM/blob/a901500618405760a564e64a6816705e29f50f9f/test/linux/slaveinfo/slaveinfo.c#L332C5-L332C15), though this could be simplified a bit.
Maybe there is a better way of getting current mapping, because the mapping is already read out when doing `config_map()` and might be available somehow, I need to check.

tomhira, 2023-08-22:
Hello bnjmnp. Thanks always.
I got the read as follows:
1C12:0x0       0x01
1C12:0x1       0x1600
1C13:0x0       0x01
1C13:0x1       0x1a00
I am seeing how mapping is done in twincat also,
so this is same with twincat, and ok.

Other codes for check are:
class RfgOutputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('field1', ctypes.c_bool),
        ('field2', ctypes.c_bool),
        ('field3', ctypes.c_uint32),
        ('field4', ctypes.c_float)
    ]

++++++++++++++++++++++++++++++
            rfg_output_data = RfgOutputPdo()
            rfg_output_data.field1 = 1
            rfg_output_data.field2 = 2
            rfg_output_data.field3 = 3
            rfg_output_data.field4 = 4
++++++++++++++++++++++++++++++

Why I am asking is:
Somehow, field 4 is never reflected or updated with this:
class RfgOutputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('field1', ctypes.c_bool),
        ('field2', ctypes.c_bool),
        ('field3', ctypes.c_uint32),
        ('field4', ctypes.c_float)
    ]
but field4 is reflected if I do: 
class RfgOutputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('field1', ctypes.c_bool),
        ('field2', ctypes.c_bool),
        #('field3', ctypes.c_uint32),
        ('field4', ctypes.c_float)
    ]
or
class RfgOutputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('field1', ctypes.c_bool),
        ('field2', ctypes.c_bool),
        ('field4', ctypes.c_float)
        ('field3', ctypes.c_uint32),
    ]

Does anybody experienced this kind of behavior?

tomhira, 2023-08-22:
Upon furthur trial, I found that the following make field3 and field4 reflected and updated, but field 2 cannot be updated:
class RfgOutputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('field1', ctypes.c_bool),
        ('field3', ctypes.c_uint8),
        ('filed4', ctypes.c_float),
        ('field2', ctypes.c_bool)
    ]

For 0x1600 0x01 is field1
                   0x02 is field2
                   0x03 is no assign
                   0x04 is field3
                   0x05 is field4
and 0x7000 0x01 is field1
                    0x02 is field2
        0x7004 0x01 is field3
        0x7005 0x01 is field4
I cannot figure out why the field2 is not updated