bnjmnp/pysoem #118: [HELP] Is there a redundancy function?

8188, 2023-09-05:
I want to read device names. Below is my codes. But when using redundancy, it can find nothing.
```
adapters = pysoem.find_adapters()
adpt = [x.name for x in adapters]
master = pysoem.Master()
for net in adpt:
    master.open(net)
    if master.config_init() > 0:
        for device in master.slaves:
            print(f"{net} : {device.name}")
master.close()
```

bnjmnp, 2023-09-10:
You are looking for the equivalent of the `ec_init_redundant()` function? So far I don't have it exposed by the PySOEM. Should not be too complicated to add this, let's see if I can do that soon.

bnjmnp, 2023-09-10:
@8188 Do you actually know what the "redport" parameter is used for?

8188, 2023-09-11:
@bnjmnp Sorry, I don't actually know that.