bnjmnp/pysoem #121: Help needed with reading SDO & Uploading SDO map

Michaelicious, 2023-10-15:
I'm having 2 issues: 
1. Reading/Writing SDO
2. Uploading XML file with SDO map / Reading object dictionary. 

I'm trying to read/write SDO's. can find a documentation on the proper way to initialize a network.
`master = ecMaster()`
`con_thread = master.ecConnection().join()                    # @threaded ---- > self.device_count = self.init_config()`
`print(master.device_count > 0)`
`if master.device_count > 0 :`
`    # s2 = master.slaves[0].dc_sync(True, 1000000)`
`    s1 = master.config_dc()`
`    s2 = master.config_map()`
`    print(s1,',',s2)`
`    od = master.slaves[0].od`

raises `SdoInfoError: Sdo List Info read failed ` due to ` od = master.slaves[0].od`

Without calling OD, Sometimes 'master.config_map()' raises 
'ConfigMapError: [MailboxError(1, 19, 'Unknown'), MailboxError(1, 19, 'Unknown'), MailboxError(1, 19, 'Unknown'), ... '

But usually I'm able to read an SDO as following:
`sdo = master.slaves[0].sdo_read(0x6060, 0)`
`sdo_decode = struct.unpack('<' + 'I' * (len(sdo) // 4), sdo)  # using struct.unpack('I' , sdo) Raised some error about 4 bits`

so 'sdo_decode' is after converting from bytes.  BUT ITS AN ETHERNET FRAME. 
how do I get SDO data from it ? what's 'master.send_processdata()' and how to use it ?
I'm stuck on it for days now, I would love some brief explanation on Read/write and setup eCat network properly.

bnjmnp, 2023-10-15:
> I'm trying to read/write SDO's. can find a documentation on the proper way to initialize a network.

One (proper) way of initializing a network is showcased in the [examples/basic_example.py](https://github.com/bnjmnp/pysoem/blob/321f2f36169553aeb6ce20c6b3b5103f22c2de0a/examples/basic_example.py), look for the `run()` function in there.
The cause for the `SdoInfoError`, I don't understand for now, but maybe you don't need it at all. Did you try to add a little delay (via `time.sleep()`) between `config_dc()` and `config_map()` to get rid of the `ConfigMapError`?

Coming to the `sdo_read()`, you probably need to do the unpack like this:
```py
sdo_decode = struct.unpack('b', sdo)
```
Object `0x6060` is probably just a two byte integer (SINT) therefore you need the `b` as format character. (You also mixed up you code with 0x6040 a bit)
Did you also read through the [Reading and Writing CoE Objects](https://pysoem.readthedocs.io/en/latest/coe_objects.html) section in the readthedocs documentation?

> 2. What's master.send_processdata() and master.recieve_processdata() ? how to use them ?

`master.send_processdata()` is used to send the PDO EtherCAT frame (that contains output PDO data) into the network, and `master.recieve_processdata()` is used to wait for frame to return (and it copies the input PDO data into a buffer). Check out this video [https://youtu.be/z2OagcHG-UU?si=0f4zRq4R60QrrOFI](https://youtu.be/z2OagcHG-UU?si=0f4zRq4R60QrrOFI), `master.send_processdata()` makes the "telegram" leave the master, and `master.recieve_processdata()` collects the retuning "telegram".

> 3. And what's up with the 4 bit error ? how do I verify I decoded the frame correctly ?

This is about the 4 byte error, we might have already solved with the `b` as format character?

> 4. Is there a way to upload an SDO map XML file ?

No. But I admit it would be nice if reading-in the object directory from the XML file would be possible, to reduce the trouble converting bytes returned by `sdo_read()` to a equivalent Python data type.

If you are about to operate a servo drive, check out [#89](https://github.com/bnjmnp/pysoem/issues/89#issuecomment-1319637926) there I point to other issues that should contain valuable information and reference code.

Michaelicious, 2023-11-05:
Sharing in case it will help anyone. 
for helping out with SDO decoding i'm saving Slave.od in a dic ordered by index value, used it in a `class CustomMaster(pysoem.master)`.
```
    def _getObjectDictionary(self,try_count=0): 
        # self.od_dic_list = []
        self.od_dic_list = []
        for slave in self.slaves:
            slave_od_dic = {}
            # print('working slave',slave, slave.od)
            try:
                od = slave.od
                print('OD Found')
            except pysoem.SdoInfoError as e:
                if try_count < 10:
                    print(f'SdoInfoError {try_count}:\t{e}')
                    try_count += 1
                    self._getObjectDictionary(try_count)
                # print(f'no SDO info for {format(slave.name)}')
                # print(f'SdoInfoError dir:\t{dir(e)}')
                # print(f'SdoInfoError getattributes:\t{e.__getattribute__.__doc__}')
                # print(f'SdoInfoError:\tmessage:{e.message}')
            else:
                print(slave.name+'Creating OD')
                for obj in od:
                    # key = obj.index
                    slave_od_dic[obj.index] = obj                    
                self.od_dic_list.append(slave_od_dic)       
```
than im using following functions to make read & write more easy.
```
import ctypes

def _choose_ctypes_object(data_type, bitsize):
    """ Returns a ctypes object based on the data type and bitsize from slave.od
        dataType  Ctype object
            1       only 2 sync object subindexes 32
            2       INT8
            3       INT16
            4       DINT / INT32
            5       UINT8
            6       UINT16
            7       UINT32
            8       None has been found
            9       String / c_char_p
            32      has enteries
            42      has enteries
"""
    data_type = int(data_type)
    bitsize = int(bitsize)
    match (data_type, bitsize):
        case (2, 8):            # INT8
            return ctypes.c_int8
        case (3, 16):           # INT16
            return ctypes.c_int16
        case (4, 32):           # DINT / INT32
            return ctypes.c_int32
        case (5, 8):            # UINT8
            return ctypes.c_uint8
        case (6, 16):           # UINT16
            return ctypes.c_uint16
        case (7, 32):           # UDINT
            return ctypes.c_uint32
        case (9, _):            # VISIBLE_STRING
            return ctypes.c_char_p
        case _:
            raise ValueError("Unsupported data type or bitsize")  
 
def _convert_from_binary(binary_data, data_type, bitsize):
    ctypes_type = _choose_ctypes_object(data_type, bitsize)
    if (data_type != 9): data = ctypes_type.from_buffer_copy(binary_data).value
    else: data = ctypes_type(binary_data).value.decode() # Decode getting rid of b'...'
    return data

def _convert_to_binary(data, data_type, bitsize):
    ctypes_type = _choose_ctypes_object(data_type, bitsize)
    binary_data = bytes(ctypes_type(data))
    return binary_data
```

closed 2023-12-05 by Michaelicious