bnjmnp/pysoem #123: Creating a link between master and slave

meganandd, 2023-11-15:
Hi there, 

I am trying to connect to an [Interface Force Measurement Solutions INF4 AD converter](https://www.interfaceforce.com/products/instrumentation/inf4-ethercat-two-three-and-four-sensor-weight-transmitter-and-indicator/) that is connected to some load cells. This particular module is specifically for EtherCAT signals. I am trying to use my laptop as the master in this network and am currently still trying to get my PC to recognize the INF4 device as a slave. I am using an Ethernet to USB converter (from McMaster) on the computer side and am able to access the network ID for that device. I then create an instance of the Master class and open the network adapter ID to the USB port. It seems to make the connection fine, but doesn't find any devices on it. 

The AD converter is powered on, but the LEDs that indicate the ECAT link/status have never turned on. Is there some step I am missing in designating my computer as the master? I've included the code (plus outputs) I've been using and any suggestions or help is appreciated. 

`
import pysoem
​
adapters = pysoem.find_adapters()
​
for i, adapter in enumerate(adapters):
    print('Adapter {}'.format(i))
    print('  {}'.format(adapter.name))
    print('  {}'.format(adapter.desc))

Adapter 0
  \Device\NPF_{7FA81AB5-CB8C-453E-9B9A-DF4259EB62DA}
  b'WAN Miniport (Network Monitor)'
Adapter 1
  \Device\NPF_{58C512C2-A8D8-4902-948D-B6ADF91A9A27}
  b'WAN Miniport (IPv6)'
Adapter 2
  \Device\NPF_{4BBAA3BD-11DE-4C82-B355-9AB1EE0F4D74}
  b'WAN Miniport (IP)'
Adapter 3
  \Device\NPF_{3DD2063A-AE46-467D-926C-E641B26A04DC}
  b'Bluetooth Device (Personal Area Network)'
Adapter 4
  \Device\NPF_{DBDCB4B1-97F0-45BD-97AC-1C186D773DEC}
  b'Intel(R) Wireless-AC 9560 160MHz'
Adapter 5
  \Device\NPF_{F6D9DBD9-11B0-4C5D-BC89-08377F7951BC}
  b'Microsoft Wi-Fi Direct Virtual Adapter #3'
Adapter 6
  \Device\NPF_Loopback
  b'Adapter for loopback traffic capture'
Adapter 7
  \Device\NPF_{9947106C-9BB9-4904-8216-358B884E2818}
  b'Realtek USB GbE Family Controller'
Adapter 8
  \Device\NPF_{A7466BA6-3E7D-45ED-BBCE-2A3B3E5EF820}
  b'Intel(R) Ethernet Connection (6) I219-LM'
master = pysoem.Master()

master.open(r'\Device\NPF_{9947106C-9BB9-4904-8216-358B884E2818}')

if master.config_init() > 0:
    for device in master.slaves:
        print(f'Found Device {device.name}')
else:
    print('no device found')

master.close()
master = pysoem.Master()
​
master.open(r'\Device\NPF_{9947106C-9BB9-4904-8216-358B884E2818}')
​
if master.config_init() > 0:
    for device in master.slaves:
        print(f'Found Device {device.name}')
else:
    print('no device found')
​
master.close()

no device found
`

Thank you!!!

bnjmnp, 2023-11-15:
It opens but no activity on the port, there is maybe something wrong with the access rights. You could try to start the console you are using, with administrator privileges. But usually this is not needed under Windows.

bnjmnp, 2023-12-21:
Hey @meganandd, did you have any success so far?