bnjmnp/pysoem #126: IAI Motor Controller (PCON-CB) - Not Getting to OP State Correctly

kfechter, 2024-01-04:
I am trying to control an IAI PCON-CB (EtherCat Model), using pysoem. I can get the device from PREOP into SAFEOP, but the device goes into a configuration error state whenever I try to go from SAFEOP to OP. Looking at the manuals I've found don't show much useful information (at least to me). 

I tried writing to the Tx and Rx PDO assign variables, however they all appear to be read-only. I've tried doing this in both pre-op and safe-op states. 

below is my example code, the only device in the chain is the PCON (Laptop -> Ethernet -> PCON)

`import sys
import pysoem
import struct
import time
import ctypes

class InputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ("current_position", ctypes.c_int32),
        ("command_current", ctypes.c_uint32),
        ("current_speed", ctypes.c_int32),
        ("alarm_code", ctypes.c_uint16),
        ("status_signal", ctypes.c_uint16),
        ("positioning_completion_signal", ctypes.c_bool),
        ("home_return_completion", ctypes.c_bool),
        ("moving_signal", ctypes.c_bool),
        ("alarm", ctypes.c_bool),
        ("operation_preperation_end", ctypes.c_bool),
        ("psfl", ctypes.c_bool),
        ("bit06_reserved_0", ctypes.c_bool),
        ("balm_alml", ctypes.c_bool),
        ("operation_mode_status", ctypes.c_bool),
        ("bit09_reserved_0", ctypes.c_bool),
        ("bit10_reserved_0", ctypes.c_bool),
        ("bit11_reserved_0", ctypes.c_bool),
        ("zone1", ctypes.c_bool),
        ("zone2", ctypes.c_bool),
        ("controller_ready", ctypes.c_bool),
        ("emergency_stop", ctypes.c_bool),
    ]

class OutputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ("target_position",ctypes.c_int32),
        ("positioning_band", ctypes.c_uint32),
        ("velocity", ctypes.c_uint16),
        ("acceleration_deceleration", ctypes.c_uint16),
        ("pressing_current_limit", ctypes.c_uint16),
        ("control_signal", ctypes.c_uint16),
        ("positioning_command", ctypes.c_bool),
        ("home_return", ctypes.c_bool),
        ("pause", ctypes.c_bool),
        ("reset", ctypes.c_bool),
        ("servo_on_command", ctypes.c_bool),
        ("jog_inch_select", ctypes.c_bool),
        ("negative_jog", ctypes.c_bool),
        ("positive_jog", ctypes.c_bool),
        ("reserved", ctypes.c_bool),
        ("gsl_0", ctypes.c_bool),
        ("gsl_1", ctypes.c_bool),
        ("push_motion_spec", ctypes.c_bool),
        ("push_direction_spec", ctypes.c_bool),
        ("operating_mode_select", ctypes.c_bool),
        ("forced_brake_release", ctypes.c_bool),
    ]

def read_sdo_info():
    # Open the SOEM Master
    master = pysoem.Master()
    master.open('enx24f5a2f189da')

    if master.config_init() > 0:
        master.config_map()
        for slave in master.slaves:
            try:
                od = slave.od
            except pysoem.SdoInfoError:
                print(f"No SDO Info for {slave.name}")
            else:
                print(slave.name)
                if slave.name == "IAI-EtherCAT":
                    for obj in od:
                        print(f" Idx: {hex(obj.index)}; Code: {obj.object_code}; Type: {obj.data_type}; BitSize: {obj.bit_length}; Access: {hex(obj.obj_access)}; Name: {obj.name}")
                        for i, entry in enumerate(obj.entries):
                            print(f'   Subindex: {i}; Type: {entry.data_type}; BitSize: {entry.bit_length}; Access: {hex(entry.obj_access)}; Name: {entry.name}')

                
                    # Writing Device Stuff
                    map_input_bytes = struct.pack("BxHHH", 3, 0x1A04, 0x1a0f, 0x1a1d)
                    map_output_bytes = struct.pack("BxHHHHHHHHHHHHHHHHHHHHH", 21, 0x160A, 0x160C, 0x161C, 0x160B, 0x160F, 0x161D, 0x1617,0x160A,0x160A,0x160A,0x160A,0x160A,0x160A,0x160A,0x160A,0x160A,0x160A,0x160A,0x160A,0x160A,0x160A)

                    slave.state = pysoem.PREOP_STATE
                    slave.write_state()

                    slave.state = pysoem.SAFEOP_STATE
                    slave.write_state()
                    
                    #slave.sdo_write(0x1c13, 0x00, map_input_bytes, True)
                    #slave.sdo_write(0x1c12, 0, map_output_bytes, True)

                 

                    # slave.state = pysoem.OP_STATE
                    # slave.write_state()

                    # output_data = OutputPdo()
                    # output_data.servo_on_command = True
                    # output_data.target_position = 20
                    # output_data.velocity = 20
                    # output_data.home_return = True
                    # output_data.control_signal = 2
                    # slave.output = bytes(output_data)

                    # master.send_processdata()
                    # master.receive_processdata(1_000)
                    # time.sleep(0.01)
                    # #data = InputPdo.from_buffer_copy(slave.input)
                    # print(slave.input)

                # output_data.modes_of_operation = modes_of_operation['Profile Velocity Mode']
                # output_data.target_velocity = 500
                # for control_cmd in [6, 7, 15]:
                #     output_data.controlword = control_cmd
                #     slave.output = bytes(output_data)
                #     master.send_processdata()
                #     master.receive_processdata(1_000)
                #     time.sleep(0.01)

                    

if __name__ == '__main__':
    print("SOEM Dump Started")
    read_sdo_info()`

kfechter, 2024-01-04:
the commented out code was other things I tried based on other examples. however I'm 99.9% sure that most of the other motor controller code won't cross over to this PCON.

bnjmnp, 2024-01-09:
Hi @kfechter,
to go into SafeOP state you should only use the `config_map()` function. If your PDO mapping is fixed this is good, less options to do something wrong.

You might want to use the code from the [minimal_example.py](https://github.com/bnjmnp/pysoem/blob/23ef331bdf30307bb778c1f65b7125b104edad08/examples/minimal_example.py) to go into OP state.

```py
            self._master.state = pysoem.OP_STATE
            self._master.write_state()

            self._master.state_check(pysoem.OP_STATE, 50000)
            if self._master.state != pysoem.OP_STATE:
                self._master.read_state()
                for slave in self._master.slaves:
                    if not slave.state == pysoem.OP_STATE:
                        print('{} did not reach OP state'.format(slave.name))
                        print('al status code {} ({})'.format(hex(slave.al_status),
                                                              pysoem.al_status_code_to_string(slave.al_status)))
                raise Exception('not all slaves reached OP state')
```

The "al status code" might give you a hint on what went wrong.

Same `state_check()` on `SAFEOP_STATE` you should actually also do after the `config_map()` function call.

kfechter, 2024-01-13:
Thank you for the reply! I will try the minimal example the next time I have access to the equipment. I know using the examples from the base SOEM, it at least seems to get to OP State so I'm most likely doing something wrong. I'm also learning ethercat at the same time so its a bit of a process.

kfechter, 2024-01-16:
definitely getting further, so that's nice. With the example, I get to safe_op properly, however during the transition to op state, the alarm status code is 0x1b (Sync Manager Watchdog). through my research it seems to have to do with dc_sync, just gotta figure out where the values come from.

kfechter, 2024-01-16:
ill close this one, since my new problem isnt related to getting to OP

closed 2024-01-16 by kfechter