bnjmnp/pysoem #131: Handle Safety

szaabe, 2024-02-09:
Hi! Could you help me is it possible that to write FSOE outputs too?

bnjmnp, 2024-02-10:
The SOEM project PySOEM is the wrapper for, does not provide any convenient functions for FSOE, this is why PySOEM does not have any support for it. FSOE needs to be implemented on a pretty low level, unfortunately this is also not possible to do on top of PySOEM.

Here are some links to the SOEM project relate to FSOE.

* [SOEM issue #132](https://github.com/OpenEtherCATsociety/SOEM/issues/132)
* [SOEM issue #257](https://github.com/OpenEtherCATsociety/SOEM/issues/257)
* [fsoe_sample.c](https://github.com/rtlabs-com/fsoe-soem-demo/blob/89cb5a926fd2310ff6c27660ecd328dc18467aec/test/linux/fsoe_sample/fsoe_sample.c)

szaabe, 2024-02-11:
Sad to hear that :( .. I'm not at home in C++ that much, so I can't afford not to learn it.Thank you for fast reply!