bnjmnp/pysoem #137: Add meaning of 'network adapter ID' to documentation

RobertoRoos, 2024-04-24:
New to `pysoem` and `soem` and I had some trouble to get started with the basics.

This page https://pysoem.readthedocs.io/en/latest/basics.html is pretty clear, except I can't find anywhere what the 'network adapter ID' looks like.
I'm on Windows, so is the ID the name as show in Control Panel? (E.g. "Ethernet 2") Or the adapter type? (E.g. "I219-V")

RobertoRoos, 2024-04-24:
I found out now by using `pysoem.find_adapters()`.
My device name is `\\Device\\NPF_{2D594793-2E69-4C58-90F4-3164E860643B}`, but I am curious where I could have found this name manually.

bnjmnp, 2024-06-30:
This is probably badly or not documented in SOEM as well.

Short term, we could at least write a comment mentioning `pysoem.find_adapters()` in the example where the `open()` function is called.

RobertoRoos, 2024-07-25:
Fixed by 148

closed 2024-07-25 by RobertoRoos