bnjmnp/pysoem #142: Open function returns always None

nzaquini, 2024-06-13:
Hi,
I'm moving my first steps in trying to connect to an EtherCAT adapter.

I started writing some basic code (see below). My problem is that the master.open(adapter.name) function returns always `None` instead of a valid boolean.

I have the same result both on Windows 11 and on the WSL virtual machine.
I don't know why open() is returning `None`. I've installed Npcap 1.79 with WinPcap Api-compatible Mode flag.

Thanks for any help,
Nicola

    # Test Pysoem Code
    import pysoem
    import time
  
    def main():
      adapters = pysoem.find_adapters()

      # Initialize the EtherCAT master
      master = pysoem.Master()
      
      for i, adapter in enumerate(adapters):
          print('Adapter {}'.format(i))
          print('  name: {}'.format(adapter.name))
          print('  description: {}'.format(adapter.desc))
  
          # Scan for slaves
          try:
              master.close()
              time.sleep(2)
              masterOpenResult = master.open(adapter.name)
              if masterOpenResult == None:
                  print('EtherCAT Master open error (value: None)')
              elif masterOpenResult:
                  print('EtherCAT Master opened successfully')
  
                  # Find and initialize all connected slaves
                  if master.config_init():
                      print(f'Found {len(master.slaves)} slaves')
  
                      # Print some basic information about each slave
                      for idx, slave in enumerate(master.slaves):
                          print(f'Slave {idx+1} - Name: {slave.name}, Output size: {slave.output_size}, Input size: {slave.input_size}')
                      
                      # Close the master
                      master.close()
                      print('EtherCAT Master closed')
                  else:
                      print('No slaves found or failed to configure')
              else:
                  print('Failed to open EtherCAT master')
                  master.find_adapters()
          except pysoem.SOEMError as e:
              print(f'SOEMError: {e}')
          except Exception as e:
              print(f'Unexpected error: {e}', file=sys.stderr)
    
    if __name__ == '__main__':
        main()

bnjmnp, 2024-06-17:
Hi,
The `open()` function is not supposed to return anything other than `None`. If an error occurs the function will raise an exception, thus you don't need to check the return value.

nzaquini, 2024-06-18:
Thank you very much.

closed 2024-06-18 by nzaquini