bnjmnp/pysoem #143: slave.write_eeprom() bug?

Michaelicious, 2024-06-19:
I'm trying to do eeprom_write, and got exceptions while trying to input bytes([1,1,1,1]), and multiple variants of it. 
So I've tried to write exactly what I read, and it still throws an exception
Couldn't find any usage examples
Any ideas ?

[image]

MartinALL91, 2024-06-19:
The input data to the eemprom_write method must be [2 bytes](https://github.com/bnjmnp/pysoem/blob/master/pysoem/pysoem.pyx#L814). The data retrieved by the eemprom_read method is [4 bytes](https://github.com/bnjmnp/pysoem/blob/master/pysoem/pysoem.pyx#L793).

Michaelicious, 2024-06-19:
> The input data to the eemprom_write method must be [2 bytes](https://github.com/bnjmnp/pysoem/blob/master/pysoem/pysoem.pyx#L814). The data retrieved by the eemprom_read method is [4 bytes](https://github.com/bnjmnp/pysoem/blob/master/pysoem/pysoem.pyx#L793).

thats why the reading is done by jumps of 2 (Meaning i cant eeprom_read 1, only 0 or 2 should work)
But eeprom_write is done 2 bytes at a time, therefore i'm able to use odd indexes ?