bnjmnp/pysoem #144: Transfer `pysoem` into the `OpenEtherCATsociety` Github group

RobertoRoos, 2024-06-21:
The title kind of says it all: could we consider transferring this repository to https://github.com/OpenEtherCATsociety, where `SOEM` itself is also hosted?

The main benefit is we wouldn't be dependent any more on a single maintainer.

@bnjmnp what do you think?
I'm currently sending an email to the OEC society to ask what they think.

bnjmnp, 2024-06-30:
Hi @RobertoRoos, yes, I'm open to do this. I'm curios on what the people from OEC say.

mekerhult, 2024-07-01:
Hi @bnjmnp and @RobertoRoos. RT-Labs are open to discuss this with you @bnjmnp. I cannot find your email address, so please send us an email (support@rt-labs.com), and we'll take it from there. Best regards, RT-Labs team.

RobertoRoos, 2024-08-22:
I don't want to press, but any progress with a potential transfer?

Maybe I should elaborate on my interest: I want to use `pysoem` for a project but I'm only allowed to do so if I'm confident in its future prooffness. Having more than a single moderator behind the project is a big part of that.