bnjmnp/pysoem #145: Code completion / Intelisense not working for

RobertoRoos, 2024-07-01:
I couldn't find a reference to this yet: it seems code completion / intelisense / etc. isn't working at all for `pysoem`:
[image]

This is in PyCharm, but the same happens in VS Code.

I imagine this is a consequence of Cython?
Maybe we could add some `.pyi` files for code analysis. Ideally generated from the Cython source files.

RobertoRoos, 2024-07-01:
I'm a little surprised code suggestions are also not working form a console, because `__dir__` looks to be implemented correctly:
[image]

StxiTestStation, 2024-08-11:
Was just about to post same issue.
installed pysoem via pip. My code runs, yet intellisense can't also find sdo_read or anything else
besides importing pysoem, the ide can't find nothing within (pysoem.XXX)
I'm also unable to import CdefSlave which I tried in order to declare type on a property.

I don't mind creating pyi file and share it. Will it help or is there a better solution that will be less manual?

Michaelicious, 2024-08-13:
Asked gpt to read the docs and create pyi file, so far it works. 
```
# .\venv\Lib\site-packages\pysoem\pysoem.pyi

from typing import List, Optional, Union

class Master:
    slaves: List["CdefSlave"]
    sdo_read_timeout: int
    sdo_write_timeout: int
    
    def close(self) -> None: ...
    def config_dc(self) -> bool: ...
    def config_init(self, usetable: bool = False) -> int: ...
    def config_map(self) -> int: ...
    def config_overlap_map(self) -> int: ...
    
    @property
    def dc_time(self) -> int: ...
    
    @property
    def expected_wkc(self) -> int: ...
    
    def open(self, ifname: str, ifname_red: Optional[str] = None) -> None: ...
    def read_state(self) -> int: ...
    def receive_processdata(self, timeout: int = 2000) -> int: ...
    def send_overlap_processdata(self) -> int: ...
    def send_processdata(self) -> int: ...
    
    @property
    def state(self) -> int: ...
    
    def state_check(self, expected_state: int, timeout: int = 50000) -> int: ...
    def write_state(self) -> int: ...

class CdefSlave:
    config_func: Optional[callable]
    name: str
    id: int
    man: int
    rev: int
    
    def dc_sync(self, act: bool, sync0_cycle_time: int, sync0_shift_time: int = 0, sync1_cycle_time: Optional[int] = None) -> None: ...
    def eeprom_read(self, word_address: int, timeout: int = 20000) -> bytes: ...
    def eeprom_write(self, word_address: int, data: bytes, timeout: int = 20000) -> None: ...
    def foe_read(self, filename: str, password: int, size: int, timeout: int = 200000) -> bytes: ...
    def foe_write(self, filename: str, password: int, data: bytes, timeout: int = 200000) -> None: ...
    def mbx_receive(self) -> int: ...
    def reconfig(self, timeout: int = 500) -> int: ...
    def recover(self, timeout: int = 500) -> int: ...
    def sdo_read(self, index: int, subindex: int, size: int = 0, ca: bool = False) -> bytes: ...
    def sdo_write(self, index: int, subindex: int, data: bytes, ca: bool = False) -> None: ...

# Additional helper functions
def find_adapters() -> List["Adapter"]: ...
def al_status_code_to_string(code: int) -> str: ...

class Adapter:
    name: str
    desc: str
```

RobertoRoos, 2024-08-19:
With v1.1.8 this is already improved. Without the asterisk import any IDE should give code completion to functions etc.
However, docs information is not processed and is not incorporated by the IDE.

I gave a `.pyi` file a try, based on the sample from @Michaelicious . I moved some doc blocks from the `.pyx` into the `.pyi` too and after installing the result is good in the IDE: code completion works in full like you'd expect from a native Python package.

However, now the docs generation through Sphinx doesn't work... Apparently Sphinx cannot work with .pyi files (yet).

I'd suggest we leave this as it is. Listing of content alone is a bit improvement.