bnjmnp/pysoem #160: HCFA Y7 servo motor is not working

RoboticsGroup2024, 2024-09-30:
Hello,
I am working on a robot that is controlled via 6 hcfa y7 servo motors,
i configured the motors to work over pysoem using PDO in position profile mode and ethercat using an incremental encoder, 
All the motors are working properly except the second one.
it goes to [8.bb] situation and do nothing while the others go to run situation and work as demanded. 
I configured all the motors withe the same configuration and verified that the configurations are written correctly by reading the corresponding registers, however it still not working although it is working using usb protocol in speed profile mode!!!

reading some registers i noticed that the hardware version of all the motors except the second one is 3718 while the hardware version for the second motor is 3413. and i noticed too, that after configuration and the motor goes to [8.bb] it goes to [A.020] after restart.

how i can fix the problem and make the second motor work?

thank you in advance

bnjmnp, 2024-10-21:
Hi @RoboticsGroup2024. This a very specific hardware issue.
I assume reaching out to the device vendors technical support you might have more success.
Was it by chance already resolved?

RoboticsGroup2024, 2024-10-21:
thank you for your reply my friend,
yes it worked correctly when we changed the software to be like the other motors,

best regard

bnjmnp, 2024-10-21:
Good to hear, so we can close this.

closed 2024-10-21 by bnjmnp

RoboticsGroup2024, 2024-10-21:
yes thank you