bnjmnp/pysoem #166: Exeptions after EEPROM writing

Michaelicious, 2024-12-18:
I've built a system in order to configure and run scripts on an EtherCAT slave
I've encountered in a situation where if I'm writing EEPROM to the slave, I getting Exceptions.
It happens only after eeprom write, if the eeprom was already written it dosent happen.
The exceptions vary, but they are usually: WkcError and SdoInfoError (i think the last one is when the slave.od was not existing)

I've tried using slave.recover() and slave.reconfig() but both returning 0.

1. Where is the object dictionary stored (is it an 0x1xxx SDO or something like that? )
It's different from CANOPEN where I must upload the eds file each time.
I just couldn't find any information regarding this so it's bothering me.

2. what should I do after EEPROM writing? 
my system has relays so I'm able to power cycle.

Code :
`
def read_eeprom_bin_file(path):
    with open(path, 'rb') as file:
        data = file.read()
        return data

def create_delete_bytearray(binfile:bytearray):
    # Define the size of the bytearray
    size = len(binfile)
    # Create a bytearray containing all 'f'
    byte_array = bytearray(b'\xFF' * size)

    return byte_array

def delete_eeprom(slave:SLAVE_TYPE):
    """Deletes, reads and verifys its 0xff..."""
      bin_content = read_eeprom_data(slave)
      if bin_content: 
          delete_content = create_delete_bytearray(bin_content)
          write_eeprom_data(delete_content,slave)
          verified = verify_eeprom_data(delete_content,slave)
          print(f"deletd eeprom: {verified}")
          return verified
    return False
    # write_eeprom_data(bin_content,slave)
            
def read_eeprom_data(slave:SLAVE_TYPE):
    EEPROM_SIZE = 1024
    # eeprom_data = []
    # stop = 0
    binr = bytearray()
    if not isinstance(slave, ethercatTIM):
        return 
    for i in range(0, EEPROM_SIZE, 2):
        # print('{:04x}:'.format(i), end='')
        
        read = slave.the_slave.eeprom_read(i)
        # binr.extend(read)
        for b in read:
            binr.append(b)
            # if b != 255: stop = 0
            # elif stop > 8: break
            # else: stop += 1
    return binr

def write_eeprom_data(bin_file,slave:SLAVE_TYPE):
    result = True
    it = iter(bin_file) # Single iterable object to bind 2 bin_file objects
    for i, (b1, b2) in enumerate(zip(it, it)):
        try: 
            slave.the_slave.eeprom_write(i, bytes([b1, b2])) # type: ignore
        except Exception as e: 
            slave.logger.error(f'eeprom write exception{e}')
            result = False
    slave.logger.info(f'eeprom write finished')
    return result

def verify_eeprom_data(eeprom_bin_file,slave:SLAVE_TYPE):
    # --- 
    eeprom_data_read = read_eeprom_data(slave)
    return eeprom_data_read == eeprom_bin_file

if master.findSlaves(): 
      # basicly doing mater.device_count = master.master.config_init()
       # the master has master.master : psoyem.Master, bad naming
            print(f"found {master.device_count} slaves")
            slave = master.custom_slaves[0]

            delete_eeprom(slave)
            slave.setPowerCycle()
            master.device_count = 0
            while not master.device_count > 0: 
                master.device_count = master.master.config_init()
                time.sleep(.5)
            print(f"found {master.device_count} slaves")
            eeprom_bin_data = read_eeprom_bin_file()
            written = write_eeprom_data(eeprom_bin_data, slave)
            print("EEPROM Written: {}".format(written))
            verified = verify_eeprom_data(eeprom_bin_data, slave)
            print("EEPROM Verified: {}".format(verified))
            time.sleep(1)
            master.master.config_init()
            TEST_NAME = "vbus" # Reads
            result = test_lib[TEST_NAME][TEST_NAME](slave)
            # Doing readSDO expects pysoem.wkcError and tryins agian after doing slave recover and reconfig` 

Exception Example: (Also there is an issue with importing the pysoem.WkcError, just FYI)
`ERROR - while reading SDO 0x6079, 0 got Exeption:
Traceback (most recent call last):
  File "c:\git\ATE_TiM\devices\Slaves.py", line 522, in readSDO
    sdo = self.the_slave.sdo_read(index, subindex) # type: ignore
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "src\\pysoem\\pysoem.pyx", line 719, in pysoem.pysoem.CdefSlave.sdo_read`

bnjmnp, 2025-01-14:
Hi @Michaelicious, did you by any chance try to increase the timeout for `eeprom_write()`? The third parameter of `eeprom_write()` is an optional timeout, which is 20000 (us) by default.

Michaelicious, 2025-01-14:
I haven't, but I dont think it was the issue as the eeprom actually gets written, and I verify it as well. 

The way I have solved it is by doing 'master.config_init()' again if eeprom was installed for the first time on one of the slaves.

It's probably something EtherCAT related, i've tried to look in SOEM code but couldn't figure it out.

closed 2025-03-10 by Michaelicious