bnjmnp/pysoem #168: how to turn on servo

ankitbarad123, 2025-01-31:
I have EtherCAT servo drive and ESI file. 
I am very new to this. I have found slave device in the program.
what are the next steps? All I want to do is turn on the servo drive. 
I have follow the basic example but it sends out many errors.

bnjmnp, 2025-02-02:
In some issues, there are code examples on how to operate a servo drive based on the DS402 drive profile:

* #30
* #33
* #43
* #70
* #73
* #76

I hope you find them helpful.

ankitbarad123, 2025-02-07:
I was able to read SDO, I am reading Encoder position. I want to read negative value. How can I do that? 
`import pysoem

def read_sdo(slave, index, subindex, data_type):
    """
    Read an SDO from an EtherCAT slave.

    :param slave: The EtherCAT slave object.
    :param index: The index of the SDO.
    :param subindex: The subindex of the SDO.
    :param data_type: The data type of the SDO (e.g., pysoem.UINT16, pysoem.UINT32).
    :return: The value of the SDO.
    """
    return slave.sdo_read(index, subindex, data_type)

def main():
    # Create an EtherCAT master
    master = pysoem.Master()
    master.open("\\Device\\NPF_{ABF1453B-31C2-47F8-B7DC-10C3C0C7D7A5}")
    # Scan for EtherCAT slaves
    if master.config_init() > 0:
        print(f"Found {master.config_init()} slaves")

        # Configure the slaves
        master.config_map()

        # Start the EtherCAT network
        master.state_check(pysoem.SAFEOP_STATE, 50000)
        master.state_check(pysoem.OP_STATE, 50000)

        # Read an SDO from the first slave
        slave = master.slaves[0]
        index = 0x6064  # Example index
        subindex = 0x00  # Example subindex
        data_type = pysoem.ECT_UNSIGNED32 # Example data type

        value = read_sdo(slave, index, subindex, data_type)
        print(f"SDO value: {value}")

    else:
        print("No slaves found")

if __name__ == "__main__":
    main()

`

Output:-
`Found 1 slaves
SDO value: b'\x18\xf7\xff\xff'
`

ankitbarad123, 2025-02-07:
> In some issues, there are code examples on how to operate a servo drive based on the DS402 drive profile:
> 
> * [send PDO incl. Index and SubIndex #30](https://github.com/bnjmnp/pysoem/issues/30)
> * [ CIA 402 engine #33](https://github.com/bnjmnp/pysoem/issues/33)
> * [A question of ESI and PDO #43](https://github.com/bnjmnp/pysoem/issues/43)
> * [Control Motor Drive using pySOEM #70](https://github.com/bnjmnp/pysoem/issues/70)
> * [Profile position mode #73](https://github.com/bnjmnp/pysoem/issues/73)
> * [Motion Control Leadshine CS3E D1008 #76](https://github.com/bnjmnp/pysoem/issues/76)
> 
> I hope you find them helpful.

ankitbarad123, 2025-02-07:
I am trying to read absolute position. I am getting binary number. I want to read values in REAL. How can I do that? 
`

 import pysoem

   def read_sdo(slave, index, subindex, data_type):

    return slave.sdo_read(index, subindex, data_type)

   def main():
 
    master = pysoem.Master()
    master.open("\\Device\\NPF_{ABF1453B-31C2-47F8-B7DC-10C3C0C7D7A5}")
  
    if master.config_init() > 0:
        print(f"Found {master.config_init()} slaves")

        master.config_map()

      
        master.state_check(pysoem.SAFEOP_STATE, 50000)
        master.state_check(pysoem.OP_STATE, 50000)

     
        slave = master.slaves[0]
        index = 0x6064  # Example index
        subindex = 0x00  # Example subindex
        data_type = pysoem.ECT_UNSIGNED32 # Example data type

        value = read_sdo(slave, index, subindex, data_type)
        print(f"SDO value: {value}")

    else:
        print("No slaves found")

    if __name__ == "__main__":
    main()

`

Output :- 
Found 1 slaves
SDO value: b'\x18\xf7\xff\xff'

ankitbarad123, 2025-02-07:
I was able to read SDO using above given reference:
Thank you @bnjmnp 
Now I will start writing the PDO

Here is a code example if it helps someone:- 
import pysoem
import struct
import time

def read_sdo(slave, index, subindex, data_type):
    """
    Read an SDO from an EtherCAT slave.

    :param slave: The EtherCAT slave object.
    :param index: The index of the SDO.
    :param subindex: The subindex of the SDO.
    :param data_type: The data type of the SDO (e.g., pysoem.UINT16, pysoem.UINT32).
    :return: The value of the SDO.
    """
    return slave.sdo_read(index, subindex, data_type)

def main():
    # Create an EtherCAT master
    master = pysoem.Master()
    master.open("\\Device\\NPF_{'adapter id}")
    # Scan for EtherCAT slaves
    master.config_init()

    # Configure the slaves
    master.config_map()

    # Start the EtherCAT network
    master.state_check(pysoem.SAFEOP_STATE, 50000)
    master.state_check(pysoem.OP_STATE, 50000)

    # Read an SDO from the first slave
    slave = master.slaves[0]

    #Read status word with Index 6041 and subindex 0
    Status_word = read_sdo(slave, 0x6041, 0x00, pysoem.ECT_UNSIGNED16)
    Status_word_val = struct.unpack('<H',Status_word)
    print("StatusWord:", str(Status_word_val[0]))
    Status_word_binary = format(Status_word_val[0], '016b')
    print(Status_word_binary)

ankitbarad123, 2025-02-10:
I am trying to run the motor in Profile Position Mode, inspired by #37 

The motor is not moving and I do get status 0*208, if I assign maximum profile velocity as per motor manual, it gives me over speed alarm. 

Below is the code, not assigning maximum profile velocity.
@bnjmnp please suggest

[image]

```
import pysoem
import ctypes
import time
import threading

pd_thread_stop_event = threading.Event()
master = pysoem.Master()
master.open("\\Device\\NPF_{ABF1453B-31C2-47F8-B7DC-10C3C0C7D7A5}")
actual_wkc = 0

def config_func():
    global device
    # software position limit
    device.sdo_write(0x607D, 1, bytes(ctypes.c_int32(0)))
    device.sdo_write(0x607D, 2, bytes(ctypes.c_int32(1000000)))
    # polarity
    device.sdo_write(0x607E, 0, bytes(ctypes.c_uint8(192)))
    # profile velocity
    device.sdo_write(0x6081, 0, bytes(ctypes.c_int32(20000)))
    # Maximum profile velocity
    #device.sdo_write(0x607F, 0, bytes(ctypes.c_int32(800000)))
    # profile acceleration
    device.sdo_write(0x6083, 0, bytes(ctypes.c_int32(50000)))
    # profile deceleration
    device.sdo_write(0x6084, 0, bytes(ctypes.c_int32(50000)))

def processdata_thread():
    global master  # not sure if this is necessary
    global pd_thread_stop_event  # not sure if this is necessary
    global actual_wkc  # not sure if this is necessary
    while not pd_thread_stop_event.is_set():
        master.send_processdata()
        actual_wkc = master.receive_processdata(10000)
        if not actual_wkc == master.expected_wkc:
            print('incorrect wkc')
        time.sleep(0.01)

if master.config_init() > 0:
    device = master.slaves[0]
    device.config_func = config_func()
    master.config_map()

    if master.state_check(pysoem.SAFEOP_STATE, 50_000) == pysoem.SAFEOP_STATE:
        master.state = pysoem.OP_STATE

        proc_thread = threading.Thread(target=processdata_thread)
        proc_thread.start()

        master.send_processdata()  # this is actually done in the "processdata_thread" - maybe it is not needed here
        master.receive_processdata(2000)  # this is actually done in the "processdata_thread" - maybe it is not needed here

        master.write_state()
        master.state_check(pysoem.OP_STATE, 5_000_000)

        if master.state == pysoem.OP_STATE:
            device.sdo_write(0x6060, 0, bytes(ctypes.c_int8(1)))  # Modes of operation - Profile position mode (pp)
            device.sdo_write(0x607A, 0, bytes(ctypes.c_int32(100000)))  # Target position

            for control_cmd in [6,7,15,31]:  # the 31 is necessary to update the "Target position", search you manual for "New set-point" - only required in pp mode
                print(hex(int.from_bytes(device.sdo_read(0x6041, 0), byteorder='little', signed=False)))
                device.sdo_write(0x6040, 0, bytes(ctypes.c_uint16(control_cmd)))  # update the controlword
                time.sleep(0.1)  # you may need to increase this sleep
            try:
                while 1:
                    time.sleep(0.1)
            except KeyboardInterrupt:
                print('stopped')
            # zero everything
            device.output = bytes(len(device.output))
            master.send_processdata()
            master.receive_processdata(1_000)

            pd_thread_stop_event.set()
            proc_thread.join()

        else:
            print('failed to got to OP_STATE')

    else:
        print('failed to got to safeop state')
    master.state = pysoem.PREOP_STATE
    master.write_state()
else:
    print('no device found')

master.close()

closed 2025-02-12 by ankitbarad123