bnjmnp/pysoem #174: pysoem for reading out of an MTS temposonic magneto-sctrictive displacement sensor

numericalfreedom, 2025-04-09:
Dear developers,

first of all thank You very much for this excellent python module for EtherCAT communication.

I am a newbie to this field and have a somewhat special and possibly trivial issue. I would like to take readings of an MTS temposonic magneto-strictive displacement sensor over EtherCAT communication. On the sensor rod, there are contactless indicator magnets and their position can be read out simultaneously with micrometer precision - great technology from space research.

With the configuration approach, I can communicate with the sensor and read out or even set individual parameters successfully over an SDO-communication.

import pysoem
master = pysoem.Master()
master.open('eno1') # I work in Debian Linux :-)
master.config_init()
print( master.slaves[0] )
...
import struct
print( struct.unpack('I', device.sdo_read(0x3101, 2))[0] )

All works great. For a single magnet on the rod, this would be sufficient, but for synchronous read out of multiple magnet positions, a PDO-communication will be possibly necessary.

For this, I understand, I have to load the protocol into the output field, send it to the slave and read out the results:

master.slaves[0].output=(???)
send_processdata()
receive_processdata()

The results are then in

master.slaves[0].input

What I unfortunately do not know, what is the command (??? above) to send to my sensor in order to trigger the protocol writing PDO-process.

Maybe, there is already some experience on this issue already.

Thank You in advance and best wishes,

Nandor
@numericalfreedom

Here the somewhat lengthy ESI-file of my sensor:

[MTS_EtherCAT_Gen_V_1.0.zip](https://github.com/user-attachments/files/19671609/MTS_EtherCAT_Gen_V_1.0.zip)

bnjmnp, 2025-04-13:
Hi @numericalfreedom 
I've no experience with this EtherCAT device, but it looks like it has a very trivial EtherCAT interface. From the ESI file one can see that no data can be send to the device. There is no object dictionary, and not `RxPdo` element, only `TxPdo`'s.
You can basically leave out the line `master.slaves[0].output=(???)`, (Py)SOEM will already create the proper (empty) PDO frame send to the device.
Like discussed in issue [#30](https://github.com/bnjmnp/pysoem/issues/30) you must map your `master.slaves[0].input` to some `class InputPdo(ctypes.Structure):` in order to access the returned sensor data.

bnjmnp, 2025-04-13:
Site note: If you place triple backticks ``` before and after code blocks, within your GitHub text, you get nicer code blocks. You can even have syntax highlighting: https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks#syntax-highlighting.

numericalfreedom, 2025-04-14:
Dear @bnjmnp ,

thank 'You for Your very kind answer and for the hints on better formatting I will gladly follow.

Your helpful work is very useful with the MTS sensor, I will review Your suggestions.

Best wishes and greeting from old Saxony,

Nandor
@numericalfreedom

numericalfreedom, 2025-04-15:
Closing the issue was premature - the system was faster than allowed :-(

numericalfreedom, 2025-04-15:
Dear @bnjmnp ,

the magneto-strictive sensors are an original invention of the company Temposonics that has been bought later by MTS systems. The measurement concept is amazing. The position of magnets on a rod are detected by sending an electromagnetic impulse into the rod and observing a returning torque wave produced at the position of magnetic detection points. As patents are free in the meantime, also other companies are using the measurement concept that is very popular for the contactless detection of piston positions in a hydraulic cylinder, for example.

The rod can be very long and multiple magnets can be detected simultaneously. We would like to employ a chained system of two sensors (ethercat slaves) with up to ten magnetic detectors each.

A solution for a simple pysoem driven data logger could maybe interest many users and it would be nice to elaborate a robust solution.

I have made my homework and most things work out very fine, but finally, the sensor returns strange values in the TxPdo input data field. I attach my current communication code and the results I get returned from the sensor when running the code (with sudo administrative rights in a virtualenv enviroment):

``` python

import pysoem
import struct
import ctypes
import time

class InputPdo(ctypes.Structure):
    _pack_ = True 
    _fields_ = [           
        ('S1' , ctypes.c_uint16),   # Status of magnet 1
        ('U1' , ctypes.c_uint32),  # Displacement of magnet 1
        ('V1' , ctypes.c_uint32),   # Velocity of magnet 1
        ('A1' , ctypes.c_uint32),   # Acceleration of magnet 1
        ('S2' , ctypes.c_uint16),
        ('U2' , ctypes.c_uint32),        
        ('V2' , ctypes.c_uint32),        
        ('A2' , ctypes.c_uint32),
        ('S3' , ctypes.c_uint16),
        ('U3' , ctypes.c_uint32),        
        ('V3' , ctypes.c_uint32),        
        ('A3' , ctypes.c_uint32),
        ('S4' , ctypes.c_uint16),
        ('U4' , ctypes.c_uint32),        
        ('V4' , ctypes.c_uint32),        
        ('A4' , ctypes.c_uint32),
    ]

master = pysoem.Master()
master.open('enp1s0')

master.config_init()
master.config_map()
#master.state = pysoem.OP_STATE
master.write_state()

device = master.slaves[0]

print()

print( "Device name = %s "   % device.name )
print( "Device type = %d"    % ctypes.c_uint32.from_buffer_copy( device.sdo_read(0x1000, 0) ).value )
print( "Error register = %d" % ctypes.c_uint8.from_buffer_copy( device.sdo_read(0x1001, 0) ).value )
print( "Vendor ID = %d"      % ctypes.c_uint32.from_buffer_copy( device.sdo_read(0x1018, 1) ).value )
print( "Product code = %d"   % ctypes.c_uint32.from_buffer_copy( device.sdo_read(0x1018, 2) ).value )
print( "Revision = %d"       % ctypes.c_uint32.from_buffer_copy( device.sdo_read(0x1018, 3) ).value )
print( "Serial number = %d"  % ctypes.c_uint32.from_buffer_copy( device.sdo_read(0x1018, 4) ).value )
print( "Smart detection = %d" % ctypes.c_uint8.from_buffer_copy( device.sdo_read(0x2000,  9) ).value )
print( "Current magnets = %d" % ctypes.c_uint8.from_buffer_copy( device.sdo_read(0x2000, 11) ).value )
print( "Maximum magnets = %d" % ctypes.c_uint8.from_buffer_copy( device.sdo_read(0x2000, 12) ).value )
print( "Cycle time = %d" % ctypes.c_uint32.from_buffer_copy( device.sdo_read(0x2000, 13) ).value )
print( "Current power supply (mV) = %d" % ctypes.c_uint32.from_buffer_copy( device.sdo_read(0x2000, 18) ).value )
print( "DC sync enabled = %d" % ctypes.c_uint32.from_buffer_copy( device.sdo_read(0x2000, 20) ).value )

master.send_processdata()
master.receive_processdata()
time.sleep( 1.0 )

print()

[... 45 lines cut ...]

U1 = 84148994
U2 = 319951120
U3 = 555753246
U4 = 791555372

The input field has a length of 56 bytes and this corresponds to the size of the input data structure, where I map the results on.

Unfortunately, the results for the status of the magnets (S) and their positions (U) are erroneous. The status (S) should be

Reserved   Magnet number  Status bit     Rest
0000000            xxxxx                      y            000

Also, the magnet positions (U) take always the same erroneous value. If I remove magnets from the rod, the "Current magnets" field reacts correctly. The test system allows for four (4) simultaneously detected magnets on the rod and this is also returned correctly.

It will be very difficult to derive conclusions from this readout - something is not working fine finally but maybe it is just a small issue. For sake of simplicity, I attach the user manual of this sensor type.

Another small question is, if You have some experience on the requirements on the ethernet adapter that can be used for ethercat connections. I have several computers, but my sensor is only ready to communicate on specific adapter and on the others I do not find any slaves.

Best wishes from sunny Saxony,

Nandor
@numericalfreedom 

[operation_manual_r-seriesv_ethercat_552059_en.pdf](https://github.com/user-attachments/files/19759880/operation_manual_r-seriesv_ethercat_552059_en.pdf)

closed 2025-04-15 by numericalfreedom

bnjmnp, 2025-04-19:
Looks like Position, Velocity, and Acceleration are signed values and should be converted using `c_int32` instead of `c_uint32`. Why the Staus is not as expected I don't understand as well. The mapping seem fine.