bnjmnp/pysoem #176: 3 similar Master Frames before Slave response

arne123, 2025-05-09:
Hi there,

I have a strange Issue, that I see 3 Master Frames before the Slave is responding on wireshark, see log below.

This is in a Windows Machine via a USB to ETH Converter.
Slave is based in a TI C2000 MCU, maybe response is not fastest.
I tried newest pysoem version and increased all timeouts by factor 10, still same.
So, maybe this is just logging Issue?

Has someone seen similar before or an Idea what I could be the issue?

Never the less Communication seems to be fine, also PDO in SAFEOP, currently I am struggling on SEFEOP -> OP transition,
during debugging that transition I have seen this Issue here.
There are always 3 Bit identical frames from the master before the slave is responding.

```
165        0.000060            Private_01:01:01            Broadcast          ECAT     29           'FPRD': Len: 1, Adp 0x1001, Ado 0x805, Wc 0
166        0.000042            Private_01:01:01            Broadcast          ECAT     29           'FPRD': Len: 1, Adp 0x1001, Ado 0x805, Wc 0
167        0.000037            Private_01:01:01            Broadcast          ECAT     29           'FPRD': Len: 1, Adp 0x1001, Ado 0x805, Wc 0
168        0.001249            03:01:01:01:01:01           Broadcast          ECAT     60           'FPRD': Len: 1, Adp 0x1001, Ado 0x805, Wc 1
169        0.000043            Private_01:01:01            Broadcast          ECAT     156          'FPWR': Len: 128, Adp 0x1001, Ado 0x1000, Wc 0  Mbx(CoE SDO Req : 'Initiate Upload' (2) Idx=0x1c00 Sub=3)
170        0.000032            Private_01:01:01            Broadcast          ECAT     156          'FPWR': Len: 128, Adp 0x1001, Ado 0x1000, Wc 0  Mbx(CoE SDO Req : 'Initiate Upload' (2) Idx=0x1c00 Sub=3)
171        0.000034            Private_01:01:01            Broadcast          ECAT     156          'FPWR': Len: 128, Adp 0x1001, Ado 0x1000, Wc 0  Mbx(CoE SDO Req : 'Initiate Upload' (2) Idx=0x1c00 Sub=3)
172        0.001256            03:01:01:01:01:01           Broadcast          ECAT     156          'FPWR': Len: 128, Adp 0x1001, Ado 0x1000, Wc 1  Mbx(CoE SDO Req : 'Initiate Upload' (2) Idx=0x1c00 Sub=3)
173        0.000040            Private_01:01:01            Broadcast          ECAT     30           'FPRD': Len: 2, Adp 0x1001, Ado 0x80d, Wc 0
174        0.000037            Private_01:01:01            Broadcast          ECAT     30           'FPRD': Len: 2, Adp 0x1001, Ado 0x80d, Wc 0
175        0.000034            Private_01:01:01            Broadcast          ECAT     30           'FPRD': Len: 2, Adp 0x1001, Ado 0x80d, Wc 0
176        0.001273            03:01:01:01:01:01           Broadcast          ECAT     60           'FPRD': Len: 2, Adp 0x1001, Ado 0x80d, Wc 1
177        0.000047            Private_01:01:01            Broadcast          ECAT     156          'FPRD': Len: 128, Adp 0x1001, Ado 0x1080, Wc 0
178        0.000043            Private_01:01:01            Broadcast          ECAT     156          'FPRD': Len: 128, Adp 0x1001, Ado 0x1080, Wc 0
179        0.000052            Private_01:01:01            Broadcast          ECAT     156          'FPRD': Len: 128, Adp 0x1001, Ado 0x1080, Wc 0
180        0.001471            03:01:01:01:01:01           Broadcast          ECAT     156          'FPRD': Len: 128, Adp 0x1001, Ado 0x1080, Wc 1  Mbx(CoE SDO Res: Scs 2)
181        0.000238            Private_01:01:01            Broadcast          ECAT     30           'FPRD': Len: 2, Adp 0x1001, Ado 0x80d, Wc 0
182        0.000126            Private_01:01:01            Broadcast          ECAT     30           'FPRD': Len: 2, Adp 0x1001, Ado 0x80d, Wc 0
183        0.000073            Private_01:01:01            Broadcast          ECAT     30           'FPRD': Len: 2, Adp 0x1001, Ado 0x80d, Wc 0
184        0.001341            03:01:01:01:01:01           Broadcast          ECAT     60           'FPRD': Len: 2, Adp 0x1001, Ado 0x80d, Wc 1
185        0.000101            Private_01:01:01            Broadcast          ECAT     29           'FPRD': Len: 1, Adp 0x1001, Ado 0x805, Wc 0
186        0.000148            Private_01:01:01            Broadcast          ECAT     29           'FPRD': Len: 1, Adp 0x1001, Ado 0x805, Wc 0
187        0.000060            Private_01:01:01            Broadcast          ECAT     29           'FPRD': Len: 1, Adp 0x1001, Ado 0x805, Wc 0
188        0.001370            03:01:01:01:01:01           Broadcast          ECAT     60           'FPRD': Len: 1, Adp 0x1001, Ado 0x805, Wc 1
189        0.000044            Private_01:01:01            Broadcast          ECAT     156          'FPWR': Len: 128, Adp 0x1001, Ado 0x1000, Wc 0  Mbx(CoE SDO Req : 'Initiate Upload' (2) Idx=0x1c12 Sub=0)
190        0.000036            Private_01:01:01            Broadcast          ECAT     156          'FPWR': Len: 128, Adp 0x1001, Ado 0x1000, Wc 0  Mbx(CoE SDO Req : 'Initiate Upload' (2) Idx=0x1c12 Sub=0)
191        0.000032            Private_01:01:01            Broadcast          ECAT     156          'FPWR': Len: 128, Adp 0x1001, Ado 0x1000, Wc 0  Mbx(CoE SDO Req : 'Initiate Upload' (2) Idx=0x1c12 Sub=0)
192        0.001295            03:01:01:01:01:01           Broadcast          ECAT     156          'FPWR': Len: 128, Adp 0x1001, Ado 0x1000, Wc 1  Mbx(CoE SDO Req : 'Initiate Upload' (2) Idx=0x1c12 Sub=0)

.....

1705     0.000144            Private_01:01:01            Broadcast          ECAT     86           'LRW': Len: 58, Adp 0x0, Ado 0x0, Wc 0
1706     0.000038            Private_01:01:01            Broadcast          ECAT     86           'LRW': Len: 58, Adp 0x0, Ado 0x0, Wc 0
1707     0.000045            Private_01:01:01            Broadcast          ECAT     86           'LRW': Len: 58, Adp 0x0, Ado 0x0, Wc 0
1708     0.001261            03:01:01:01:01:01           Broadcast          ECAT     86           'LRW': Len: 58, Adp 0x0, Ado 0x0, Wc 3
1709     0.104021            Private_01:01:01            Broadcast          ECAT     30           'BRD': Len: 2, Adp 0x0, Ado 0x130, Wc 0
1710     0.000074            Private_01:01:01            Broadcast          ECAT     30           'BRD': Len: 2, Adp 0x0, Ado 0x130, Wc 0
1711     0.000074            Private_01:01:01            Broadcast          ECAT     30           'BRD': Len: 2, Adp 0x0, Ado 0x130, Wc 0
1712     0.001438            03:01:01:01:01:01           Broadcast          ECAT     60           'BRD': Len: 2, Adp 0x1, Ado 0x130, Wc 1
1713     0.000210            Private_01:01:01            Broadcast          ECAT     86           'LRW': Len: 58, Adp 0x0, Ado 0x0, Wc 0
1714     0.000058            Private_01:01:01            Broadcast          ECAT     86           'LRW': Len: 58, Adp 0x0, Ado 0x0, Wc 0
1715     0.000041            Private_01:01:01            Broadcast          ECAT     86           'LRW': Len: 58, Adp 0x0, Ado 0x0, Wc 0
1716     0.001394            03:01:01:01:01:01           Broadcast          ECAT     86           'LRW': Len: 58, Adp 0x0, Ado 0x0, Wc 3

```

bnjmnp, 2025-05-11:
Hi @arne123, I noticed similar things on Windows at some point. Probably also heard it from others as well.
As you already assumed, this is just a logging issue and not related to PySOEM or SOEM.

I'm a bit lazy and present you the OpenAI/ChatGPT explanation:
> Yes, it's quite possible that the confusion in frame order you're seeing in Wireshark is due to timestamp issues. Here are the main reasons this can happen, particularly on Windows:
> ### 1. Timestamp Accuracy on Windows
>
> Wireshark relies on the operating system and network drivers to provide accurate timestamps. On Windows:
>
>    The default capture mechanism (Npcap or previously WinPcap) may introduce inaccuracies or jitter in timestamps.
>
>    Windows is not a real-time OS, so timestamps might not reflect the exact time a packet was sent or received, especially under CPU load.
>
> ###  2. Delayed Packet Delivery to Wireshark
>
> Sometimes packets are received by the NIC but not immediately handed off to Wireshark for timestamping due to:
> 
>    Buffering by the driver or OS.
>
>    Interrupt moderation settings (on NIC).
>    This causes received packets to appear "later" than they really arrived.
>
> ###  3. Out-of-order Processing
>
> Even if packets arrive in order:
>
>    Windows might process and timestamp them out of order, especially if multiple CPU cores handle different parts of the network stack.
>
> ### 4. Frame Direction Confusion
>
> If you're monitoring both sent and received traffic:
>
>    Frames may appear to be received before they were sent due to the timestamping inconsistencies.
>
>    Especially true for loopback traffic or local network monitoring without a tap or mirror port.

arne123, 2025-05-13:
Hi @bnjmnp,
thanks for clarification, helped a lot!

closed 2025-05-15 by arne123