bnjmnp/pysoem #179: Lichuan LC10E communication problems

Piotrek12332121, 2025-05-25:
Hello

I am trying to control a Lichuan LC10E ( sometimes refered as A5-E) servo motor drive in Periodic synchronization position mode, however i encounter problems and cannot do so. 
During operation of the program, the drive displays on its biult-in monitor state " 88rY" which suggests that it is in Running status in desired mode and awaits servo enable signal from host computer.

```
import pysoem
import ctypes
import time
import threading
import struct
from src.helper_functions import decode_control_word, decode_status_word

PORT_NAME= '\\Device\\NPF_{1BB2F936-6269-421F-ACD7-0A0EB579544D}'  # Replace with your actual port name
#PORT_NAME="eno1"

pd_thread_stop_event = threading.Event()
master = pysoem.Master()

master.open(PORT_NAME)
actual_wkc = 0

def config_func():
    device = master.slaves[0]

    # RPDO: Control Word, Target Position
    device.sdo_write(0x1600, 0, (0).to_bytes(1, 'little'))
    device.sdo_write(0x1600, 1, (0x60400010).to_bytes(4, 'little'))  # Control Word
    device.sdo_write(0x1600, 2, (0x607A0020).to_bytes(4, 'little'))  # Target Position
    device.sdo_write(0x1600, 0, (2).to_bytes(1, 'little'))

    # TPDO: Status Word, Actual Position
    device.sdo_write(0x1A00, 0, (0).to_bytes(1, 'little'))
    device.sdo_write(0x1A00, 1, (0x60410010).to_bytes(4, 'little'))  # Status Word
    device.sdo_write(0x1A00, 2, (0x60640020).to_bytes(4, 'little'))  # Actual Position
    device.sdo_write(0x1A00, 0, (2).to_bytes(1, 'little'))

    # Assign PDOs
    device.sdo_write(0x1C12, 0, (0).to_bytes(1, 'little'))
    device.sdo_write(0x1C12, 1, (0x1600).to_bytes(2, 'little'))
    device.sdo_write(0x1C12, 0, (1).to_bytes(1, 'little'))

    device.sdo_write(0x1C13, 0, (0).to_bytes(1, 'little'))
    device.sdo_write(0x1C13, 1, (0x1A00).to_bytes(2, 'little'))
    device.sdo_write(0x1C13, 0, (1).to_bytes(1, 'little'))

    # Set mode to EtherCAT and CSP
    device.sdo_write(0x2002, 1, (9).to_bytes(1, 'little'))  # EtherCAT mode
    device.sdo_write(0x6060, 0, (8).to_bytes(1, 'little'))  # CSP mode

def processdata_thread():

    while not pd_thread_stop_event.is_set():

        target_pos = 100000
        # Read current output as mutable bytearray
        out_data = bytearray(device.output)

        # Modify control word and target position
        out_data[0:2] = struct.pack('<H', 0x000F)         # Control Word
        out_data[2:6] = struct.pack('<i', target_pos)         # Target Position

        # Assign back to device.output
        device.output = bytes(out_data)

        master.send_processdata()
        actual_wkc = master.receive_processdata(1000)
        if not actual_wkc == master.expected_wkc:
            print('incorrect wkc')

        time.sleep(0.01)

if master.config_init() > 0:
    device = master.slaves[0]
    device.config_func = config_func()
    master.config_map()

    if master.state_check(pysoem.SAFEOP_STATE, 50_000) == pysoem.SAFEOP_STATE:
        master.state = pysoem.OP_STATE

        master.send_processdata()  
        master.receive_processdata(2000)  

        master.write_state()
        master.state_check(pysoem.OP_STATE, 5_000_000)

        proc_thread = threading.Thread(target=processdata_thread)
        proc_thread.start()

        if master.state == pysoem.OP_STATE:

            print("🔧 Resetting fault (0x0080)...")
            device.sdo_write(0x6040, 0, (0x0080).to_bytes(2, 'little'))
            time.sleep(0.1)

            print("🧩 Sending Shutdown (0x0006)...")
            device.sdo_write(0x6040, 0, (0x0006).to_bytes(2, 'little'))
            time.sleep(0.1)

            print("🟡 Sending Switch On (0x0007)...")
            device.sdo_write(0x6040, 0, (0x0007).to_bytes(2, 'little'))
            time.sleep(0.1)

            # Step 4: Enable Operation
            print("🟢 Sending Enable Operation (0x000F)...")
            device.sdo_write(0x6040, 0, (0x000F).to_bytes(2, 'little'))
            time.sleep(0.1)

            decode_status_word(device.sdo_read(0x6041, 0))
            time.sleep(0.1)
            decode_control_word(device.sdo_read(0x6040,0))

            print("🟡 Sending Switch On (0x0007)...")
            device.sdo_write(0x6040, 0, (0x0007).to_bytes(2, 'little'))
            time.sleep(0.1)

            decode_status_word(device.sdo_read(0x6041, 0))
            time.sleep(0.1)
            decode_control_word(device.sdo_read(0x6040,0))

            try:
                while 1:
                    time.sleep(0.1)
            except KeyboardInterrupt:
                print('stopped')
            # zero everything
            
            device.output = bytes(len(device.output))
            master.send_processdata()
            master.receive_processdata(1_000)

            pd_thread_stop_event.set()
            proc_thread.join()

        else:
            print('failed to got to OP_STATE')

    else:
        print('failed to got to safeop state')
    master.state = pysoem.PREOP_STATE
    master.write_state()
else:
    print('no device found')

master.close()
```

After inicialization Status Word is in state : 0x0237 (0000001000110111) and  Control Word: 0x000F (0000000000001111) and I cannot proceed further into motor operation. 

[image]

What may cause this behaviour? Is there a problem with assigning PDOs and periodic communication or did I steer the CIA402 state machine  wrongly? Maybe time contriains are a factor? 

[image]
In addition to that, my slave info is as follows:
Slave 0:
  Name       : EtherCAT Driver
  Vendor ID  : 0x766
  Product Code: 0x402

ESI file and docs: 
[lc10e_manual.pdf](https://github.com/user-attachments/files/20430414/lc10e_manual.pdf)
[ESI.zip](https://github.com/user-attachments/files/20430443/ESI.zip)

marcDuboc, 2025-06-02:
Looking at your code, it seems that you are missing the DC SYNC configuration. I am not sure if it will solve your problem but for cyclic operation it is necessary. 

- in PRE OP: you should add `master.config_dc()` 
- in SAFE OP: `slave.dc_sync(act=True, sync0_cycle_time=1_000_000)` 

some servo drive require to have them both set in PRE OP.

Also `it is in Running status in desired mode and awaits servo enable signal from host computer.` yes some servo require S_ON signal  Usually it is done by assigning a pin on the DIO (CN1 or other) a pull it to 24v.

TCczs-6, 2025-08-19:
Hello, have you solved it? I am also using Lichuan LC10E and have encountered the same problem as you, which has been stuck in the 'ry' state and cannot run. If it is solved, could you share the source code? Thank you.