bnjmnp/pysoem #183: TxPDO,RxPDO size have been oddly swapped, wkc=0 and slave.inputs all zero

whichcode, 2025-09-09:
I'm working on a ethercat device which works fine at TwinCAT. When reading TxPDO data from this device in pysoem,TxPDO,RxPDO size have been oddly swapped, wkc=0 and slave.inputs all zero. TxPDO should be 125 bytes, RxPDO should be 16 bytes. But actual slave.inpput is 16 bytes. This device has a fixed PDO map, and 1C13,1C12 map were checked by report1c13,report 1c12, It seems nothing wrong, The device is driven by AX58100 SPI interface and I've tried different firmware, for examples, to reduce TxPDO,RxPDO to 4 bytes or to make PDO map changeable. This strange swap situatin is still there. Reading data by slave.sdo_read(0x6000, 1) is always correct.
Below is an excerpt of the output:
-------------
0x1C13:0 = 1
  assigned TxPDO 1: 0x1A00
    0x6000:1  (32 bits)
    0x6000:2  (32 bits)
    0x6000:3  (32 bits)
    0x6000:4  (32 bits)
   .........
0x1C12:0 = 1
  assigned RxPDO 1: 0x1600
    0x7000:1  (1 bits)
    0x7000:2  (1 bits)
    0x7000:3  (1 bits)
    0x7000:4  (1 bits)
    0x0000:0  (4 bits)
   .........
sent = 1
wkc =    0,len=   16, cycle=00 ch1=0x00000000 ch2=0x00000000
-------------
import pysoem
import struct
import time
import ctypes

IFACE = "\\Device\\NPF_{2DEDE98C-FF93-4A94-8E9E-8826BAFFBD89}"  # <-- change to your NIC

import struct

def rd(slv, idx, sub=0):
    return slv.sdo_read(idx, subindex=sub)

def report1c13(s):
        # Number of assigned TxPDOs:
        num = rd(s, 0x1C13, 0)[0]
        print('--------------------------')
        print('0x1C13:0 =', num)
        for k in range(1, num+1):
            pdo_idx = int.from_bytes(rd(s, 0x1C13, k), 'little')
            print(f'  assigned TxPDO {k}: 0x{pdo_idx:04X}')
            # Each 0x1Axx mapping: sub0 = n entries; each entry is 32-bit (idx<<16 | sub<<8 | bitlen)
            nmap = rd(s, pdo_idx, 0)[0]
            for e in range(1, nmap+1):
                entry = int.from_bytes(rd(s, pdo_idx, e), 'little')
                idx  = (entry >> 16) & 0xFFFF
                sub  = (entry >> 8) & 0xFF
                bits = entry & 0xFF
                print(f'    0x{idx:04X}:{sub}  ({bits} bits)')

def report1c12(s):
        # Number of assigned TxPDOs:
        num = rd(s, 0x1C12, 0)[0]
        print('--------------------------')
        print('0x1C12:0 =', num)
        for k in range(1, num+1):
            pdo_idx = int.from_bytes(rd(s, 0x1C12, k), 'little')
            print(f'  assigned RxPDO {k}: 0x{pdo_idx:04X}')
            # Each 0x1Axx mapping: sub0 = n entries; each entry is 32-bit (idx<<16 | sub<<8 | bitlen)
            nmap = rd(s, pdo_idx, 0)[0]
            for e in range(1, nmap+1):
                entry = int.from_bytes(rd(s, pdo_idx, e), 'little')
                idx  = (entry >> 16) & 0xFFFF
                sub  = (entry >> 8) & 0xFF
                bits = entry & 0xFF
                print(f'    0x{idx:04X}:{sub}  ({bits} bits)')   

def rd_u32(buf, off):      return struct.unpack_from("<I", buf, off)[0]
def rd_float(buf, off):      return struct.unpack_from("<f", buf, off)[0]
    
def main():
    master = pysoem.Master()
    master.open(IFACE)

    try:
        if master.config_init(True) <= 0:
            print("No slaves found")
            return

        slave = master.slaves[0]
        print(f"Configuring PDOs for: {slave.name}")
    
        master.config_map()

        report1c13(slave)
        report1c12(slave)
   
        master.state = pysoem.OP_STATE
        master.write_state()
        master.state_check(pysoem.OP_STATE, 50000)
            
        for i in range(15):

            sent = master.send_processdata()
            print(f"sent = {sent}")
            
            wkc = master.receive_processdata(2000)  # WKC = 0 or very low → PD isn’t being exchanged.

            length_r = len(slave.input);

            ch1 = rd_u32(slave.input, 0)    # Status word
            ch2 = rd_u32(slave.input, 4)    # Actual position
            print(f"wkc = {wkc:4d},len= {length_r:4d}, cycle={i:02d} ch1=0x{ch1:08X} ch2=0x{ch2:08X}")

            #ch1 = rd_float(slave.sdo_read(0x6000, 1),0)  # example index/subindex—use your device’s object dictionary
            #ch2 = rd_float(slave.sdo_read(0x6000, 2),0) # example index/subindex—use your device’s object dictionary
            #print(f"sdo cycle={i:02d} ch1={ch1} ch2={ch2}")
            time.sleep(0.1)

    finally:
        # Leave OP cleanly
        #master.state = pysoem.INIT_STATE
        #master.write_state()
        master.close()

if __name__ == "__main__":
    main()

whichcode, 2025-09-09:
by using changable PDO mapping , WKC =1 , but reading still all zero
-------------
TX_FLOAT_INDEX   = 0x6000
TX_FLOAT_SUBIDX1 = 0x01 
TX_FLOAT_SUBIDX2 = 0x02 

RX_BOOL_INDEX    = 0x7000  
RX_BOOL_SUBIDX1  = 0x01   
RX_BOOL_SUBIDX2  = 0x02 

BOOL_BITLEN = 1        
# ======================================

def u8(x):  return struct.pack('<B',  x & 0xFF)
def u16(x): return struct.pack('<H',  x & 0xFFFF)
def u32(x): return struct.pack('<I',  x & 0xFFFFFFFF)

def sdo_write(slave, idx, sub, data):
    try:
        slave.sdo_write(idx, sub, data)
    except pysoem.SdoAborted as e:
        print(f"SDO abort @ {idx:#06x}:{sub:#04x}, code=0x{e.abort_code:08X}")
        raise

def map_txpdo_two_floats(slave):
    sdo_write(slave, 0x1C13, 0, u8(0))
    sdo_write(slave, 0x1A00, 0, u8(0))
    mapping1 = (TX_FLOAT_INDEX << 16) | (TX_FLOAT_SUBIDX1 << 8) | 32
    sdo_write(slave, 0x1A00, 1, u32(mapping1))
    mapping2 = (TX_FLOAT_INDEX << 16) | (TX_FLOAT_SUBIDX2 << 8) | 32
    sdo_write(slave, 0x1A00, 2, u32(mapping2))
    sdo_write(slave, 0x1A00, 0, u8(2))
    sdo_write(slave, 0x1C13, 1, u16(0x1A00))
    sdo_write(slave, 0x1C13, 0, u8(1))

def map_rxpdo_two_bools(slave):
    sdo_write(slave, 0x1C12, 0, u8(0))
    sdo_write(slave, 0x1600, 0, u8(0))
    mapping1 = (RX_BOOL_INDEX << 16) | (RX_BOOL_SUBIDX1 << 8) | BOOL_BITLEN
    sdo_write(slave, 0x1600, 1, u32(mapping1))
    mapping2 = (RX_BOOL_INDEX << 16) | (RX_BOOL_SUBIDX2 << 8) | BOOL_BITLEN
    sdo_write(slave, 0x1600, 2, u32(mapping2))
    sdo_write(slave, 0x1600, 0, u8(2))
    sdo_write(slave, 0x1C12, 1, u16(0x1600))
    sdo_write(slave, 0x1C12, 0, u8(1))

addingding, 2025-12-09:
same to #188